bambulab/BambuStudio #2585: flow calibration - outside of the print area on A1 Mini

xxzvxx, 2023-09-23:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

When choosing A1 mini the flow calibration outside of the printing area.

### Printer

A1 Mini

### How to reproduce

slice any stl and you will notice the flow calibration lines are on  the edge of the plate when you slice the plate.

### Actual results

n/a

### Expected results

flow lines are outside of the printing area
[image]

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2023-09-25:
It is not a bug.After consulting with the R&D team of the design, it was found that the calibration line was placed outside the printing area, and the suspended part was easy to tear by hand

xxzvxx, 2023-09-25:
That's make sense to move it to outside of the printing area, but if you move it by +10mm in X direction it will looks much more professional.

sanguinaria-canadensis, 2023-09-25:
> That's make sense to move it to outside of the printing area, but if you move it by +10mm in X direction it will looks more professional.

It is intended. The starting point is outside the plate because that will be more convenion for users to pick up the tip and remove it .

closed 2023-09-26 by xxzvxx