bambulab/BambuStudio #2586: Unable to build BambuStudio Dependencies

Devinwhite, 2023-09-23:
### Bambu Studio Version

1.7.6

### Where is the application from?

Bambu Lab github releases

### OS version

Microsoft Windows 11 Pro 64-bit Ver.2009(OS build 22621.2283)

### Additional system information

OS:                           Microsoft Windows 11 Pro 64-bit Ver.2009(OS build 22621.2283)
Windows Product Key:          WFG6P
HDI Build:                    ZS7-17Q1KM4-SF6 / 15.TC0.L11.11.0.10
BIOS Version:                 E17Q1IWS.113
BIOS Release Date:            2023/07/04
EC Version:                   17Q1IWS1.10C0613202314:39:48
CPU:                          12th Gen Intel(R) Core(TM) i7-12800HX
Memory:                       32 GB @ 2000 MHz
                               - 16 GB, DDR5-4000, Samsung M425R2GA3BB0-CQKOL  
                               - 16 GB, DDR5-4000, Samsung M425R2GA3BB0-CQKOL  
Graphics:                     DisplayLink USB Device, 0 MB
Graphics:                     DisplayLink USB Device, 0 MB
Graphics:                     NVIDIA RTX A3000 12GB Laptop GPU, 12288 MB
Graphics:                     Intel(R) UHD Graphics, 1024 MB
VBIOS Version:                94.04.81.00.2b
Disk:                         SSD, SAMSUNG MZVL21T0HCLR-00B00, 953.87 GB
Disk:                         SSD, WD_BLACK SN850X 4000GB, 3726.02 GB
Network:                      Killer(R) Wi-Fi 6E AX1675i 160MHz Wireless Network Adapter (211NGW)
Network:                      Killer E3100G 2.5 Gigabit Ethernet Controller
Network:                      Bluetooth Device (Personal Area Network)
Network:                      DisplayLink Network Adapter NCM
Win32 SHIFT mode:             Unsupportive
Power Plan:                   Power Scheme GUID: 381b4222-f694-41f0-9685-ff5bb260df2e  (Balanced)
UWP User Scenario:            Extreme Performance, Performance: Hight, Fan: Auto
Graphic Switch:               Unsupportive
Security Level 2:             Supportive

### Printer

Bambu Labs X1C

### How to reproduce

1. follow the instructions in https://github.com/bambulab/BambuStudio/blob/master/doc/How%20to%20build%20-%20Windows.md
2. run commands as specified cmake ../ -G "Visual Studio 16 2019" -DDESTDIR="D:/work/Projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release
3. msbuild /m ALL_BUILD.vcxproj

### Actual results

    "D:\work\Projects\BambuStudio\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "D:\work\Projects\BambuStudio\deps\build\deps.vcxproj" (default target) (3) ->
       "D:\work\Projects\BambuStudio\deps\build\dep_OpenVDB.vcxproj" (default target) (15) ->
       "D:\work\Projects\BambuStudio\deps\build\dep_OpenEXR.vcxproj" (default target) (23) ->
       (CustomBuild target) ->
         CUSTOMBUILD : error : corrupt patch at line 5 [D:\work\Projects\BambuStudio\deps\build\dep_OpenEXR.vcxproj]
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(249,5): error MSB8066: Custom build for 'D:\work\Projects\BambuStudio\deps\build\
       CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-mkdir.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-download.rule;D:\work\Projects
       \BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-update.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-patch
       .rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-configure.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da
       1b2e230\dep_OpenEXR-build.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-build_debug.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFile
       s\3bc1d761a2014b163123587da1b2e230\dep_OpenEXR-install.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\68c521bc805cc530064eef33e47b39d2\dep_OpenEXR-complete.rule;D:\work\Projects\BambuS
       tudio\deps\build\CMakeFiles\258fc9b4ab6c35b7f3020ecde013b882\dep_OpenEXR.rule;D:\work\Projects\BambuStudio\deps\CMakeLists.txt' exited with code 128. [D:\work\Projects\BambuStudio\deps\build\de
       p_OpenEXR.vcxproj]

       "D:\work\Projects\BambuStudio\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "D:\work\Projects\BambuStudio\deps\build\deps.vcxproj" (default target) (3) ->
       "D:\work\Projects\BambuStudio\deps\build\dep_wxWidgets.vcxproj" (default target) (17) ->
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(249,5): error MSB8066: Custom build for 'D:\work\Projects\BambuStudio\deps\build\
       CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-mkdir.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-download.rule;D:\work\Proj
       ects\BambuStudio\deps\build\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-update.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidge
       ts-patch.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-configure.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\a00868e1beae331
       0baf14ddff3dd8d72\dep_wxWidgets-build.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-build_debug.rule;D:\work\Projects\BambuStudio\deps\b
       uild\CMakeFiles\a00868e1beae3310baf14ddff3dd8d72\dep_wxWidgets-install.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\68c521bc805cc530064eef33e47b39d2\dep_wxWidgets-complete.rule;D:\wo
       rk\Projects\BambuStudio\deps\build\CMakeFiles\258fc9b4ab6c35b7f3020ecde013b882\dep_wxWidgets.rule;D:\work\Projects\BambuStudio\deps\CMakeLists.txt' exited with code 1. [D:\work\Projects\BambuSt
       udio\deps\build\dep_wxWidgets.vcxproj]

    1452 Warning(s)
    3 Error(s)

### Expected results

The expected result is the dependencies would build, and I would be able to proceed with the building of the application.

### Project file & Debug log uploads

No project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Devinwhite, 2023-09-25:
7>CUSTOMBUILD : error : patch failed: src/osx/cocoa/settings.mm:224
7>CUSTOMBUILD : error : src/osx/cocoa/settings.mm: patch does not apply
7>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-mkdir.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-download.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-update.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-patch.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-configure.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-build.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-build_debug.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\e18a78d70448d9e18ed7da1648b635eb\dep_wxWidgets-install.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\823960a7111b9f21ba44d232e6412e75\dep_wxWidgets-complete.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\258fc9b4ab6c35b7f3020ecde013b882\dep_wxWidgets.rule' exited with code 1.
7>Done building project "dep_wxWidgets.vcxproj" -- FAILED.

lanewei120, 2023-10-08:
@Devinwhite 
hello, sorry for the late replay
do you still meet this issue, 
if yes, could you show the full log of the building?

and is there any error when the first cmake command executed

tsmith35, 2024-01-08:
> sometimes the building will fail due to the network downloading, we need to cleanup the patches of dep_wxWidgets before you build again you can try following commands: `cd BambuStudio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets` `git reset --hard`

@lanewei120, just wanted to say that this helped me with the same issue. Thank you!

lanewei120, 2024-01-09:
> @lanewei120, just wanted to say that this helped me with the same issue. Thank you!

@tsmith35 

Glad to hear that

thanks for the feedback~

lanewei120, 2024-06-26:
should be fixed, please try the newest codes

closed 2024-06-26 by lanewei120