bambulab/BambuStudio #2589: Manual flowcalibration reports invalid k value

ChrisVorhauer, 2023-09-23:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2 x64

### Printer

Bambu Lab P1S

### How to reproduce

Go to Main Menu "Kalibration" (Calibration, german language, region Europe) and under "Kalibrierung der Flußdynamik" (Calibration of the Fluiddynamic) choose "Manuelle Kalibraton" (Manual Calibration)

### Actual results

As long as you don't change the default k-values and intervall "Weiter" results in an error message

### Expected results

The default values for k ignore the region and use periods to separate fractions, instead of commas
As soon as you enter comma separated values you can proceed

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12706831/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-10-11:
@ChrisVorhauer  It has been fixed in V1.7.7, could you have a try

ChrisVorhauer, 2023-10-11:
No more error, but the default values are still not comma separated. So its not obvious whether to use dot or comma to change the values

closed 2023-10-11 by ChrisVorhauer