bambulab/BambuStudio #2590: Bridging issue on color transition layers

john-rubin, 2023-09-23:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Ventura 13.5.2

### Additional system information

[image]

### Printer

X1 Carbon

### How to reproduce

Slice any file with a color change on a bridge:

[image]

### Actual results

Objects with a color transition on a bridge treat the first layer of color change as requiring walls.  Objects with no color transition on a layer with a bridge do not have walls.

### Expected results

First layer should not require walls on a color change.

### Project file & Debug log uploads

[Commander Deck Box.3mf.zip](https://github.com/bambulab/BambuStudio/files/12707186/Commander.Deck.Box.3mf.zip)

Please refer to plate 5 to see my tests with this particular object.  The color transition on layer 17 bridges incorrectly.  On the part with no transition, the bridge is correct.

### Checklist of files to include

- [ ] Log file
- [X] Project file

john-rubin, 2023-09-23:
I should add this occurs regardless of wall generator settings, thin wall detection, or any change related to bridging.  It is ever-present, and only occurs when a color transition coincides with a bridge layer.

zhimin-zeng-bambulab, 2023-10-08:
Currently, due to the limitations of the coloring mechanism, if we want to modify it, the changes will be large and need to be considered as a whole. As a feature, we will schedule it in the future.

Mitra98t, 2024-02-10:
To mitigate the problem I'm currently using a workaround including a cube at the same height as the bridge and grouping it with the primary part, simply setting the filament of the cube solves the issue. Not the prettiest solution but it works, hope this helps.
[image]
[image]
[image]