bambulab/BambuStudio #2591: Show Black Filament as Dark Gray

aherta, 2023-09-23:
**Is your feature request related to a problem? Please describe.**
It's difficult to see the features of a model when using black filament because it's just solid black.

**Describe the solution you'd like**
Making black a dark gray that allows some definition to be seen would be great.

**Describe alternatives you've considered**
Printing in other colors works much better, but I use a lot of black.

**Additional context**
[image]

QingZhangBambu, 2023-09-25:
#2603

Haidiye00, 2023-09-25:
Thank you. We have fixed this issue in the upcoming V1.8 version. Stay tuned

JakeAi, 2023-11-03:
Thank god. Came here to post this same issue.

closed 2024-01-16 by Haidiye00