bambulab/BambuStudio #2594: Auto Refill dialog is not resizable

gsingh93, 2023-09-24:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.5

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to the device page
2. Click on "Auto Refill" by the AMS settings
3. A non-resizable dialog with the text cutoff is shown

### Actual results

[image]

### Expected results

I should be able to read all the text, I'm not sure what this message is trying to say about which spools are compatible.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-09-25:
Hi. @gsingh93 
I'm very sorry, we will work on optimizing it.

bam-winterfell, 2023-10-07:
The optimization has been completed and will be released in the next version. Please update promptly. Wishing you a happy life.

closed 2023-12-06 by bam-winterfell