bambulab/BambuStudio #2595: Linux Wayland support

officiallymarky, 2023-09-24:
I am using the App Image for Linux but it runs in xwayland mode (Wayland compatibility) which runs really poorly. 

Most Linux apps using electron can forced into Wayland mode eliminating the graphic and performance glitches but I can’t find anyway to force BS to use Wayland.  

Linux distributions are moving away from X11 and only supporting Wayland going forward (Fedora just recently proposed this).

xux1217, 2023-09-27:
yeah, we need wayland support directly, please add it, thanks~

hadess, 2024-01-07:
This would at least need:
- removing the `GDK_BACKEND` setting in `src/BambuStudio.cpp`
- making sure to enable Wayland support in GLFW, glew and wxWidgets

Botspot, 2024-08-05:
This is especially necessary on low-power computing platforms such as the Raspberry Pi, which now use Wayland.
Any 3D panning that is rendered through xWayland can be expected to have a low frame rate and keep the CPU occupied at 100% utilization.

I'm interested in adding BS to [our ARM app store](https://github.com/Botspot/pi-apps), alongside Cura, PrusaSlicer, and other slicers. But it needs to be usable. Maybe it is usable on xwayland, but it would be a lot more usable if wayland could be enabled.

different-name, 2024-11-02:
The prepare and preview windows do not appear on Wayland for me

Hyprland, built from branch v0.44.1-b at commit 4520b30d498daca8079365bdb909a8dea38e8d55

[image]

officiallymarky, 2024-11-02:
> The prepare and preview windows do not appear on Wayland for me
> 
> Running through xwayland, Hyprland, built from branch v0.44.1-b at commit 4520b30d498daca8079365bdb909a8dea38e8d55
> 
> ![image](https://private-user-images.githubusercontent.com/49257026/382491910-c1a93e63-61a8-40b1-8cac-cb4947ebe733.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA1ODMzNTQsIm5iZiI6MTczMDU4MzA1NCwicGF0aCI6Ii80OTI1NzAyNi8zODI0OTE5MTAtYzFhOTNlNjMtNjFhOC00MGIxLThjYWMtY2I0OTQ3ZWJlNzMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAyVDIxMzA1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2MTYyYjdlY2FiZWQ2YTI0M2Y3ODAxMzAwMmZjNGY4ZWZlYTg5ZjJmN2RmNDM0ZGZlZmYwYzcwNTRjZDVlNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.dMYZ7xDYevoJvk4qHr9ItVW1hbg5zEZzuAyuSClK1tA)

Give this a try

WEBKIT_DISABLE_DMABUF_RENDERER=1

different-name, 2024-11-03:
> Give this a try
> 
> WEBKIT_DISABLE_DMABUF_RENDERER=1

Ran `WEBKIT_DISABLE_DMABUF_RENDERER=1 bambu-studio` with no change

mordicuac, 2024-11-11:
Same issue here with an Nvidia 2060

different-name, 2024-11-12:
> The prepare and preview windows do not appear on Wayland for me

> Same issue here with an Nvidia 2060

There's an issue open for the lacking 3D view on Wayland with Nvidia 560 and 565 drivers

https://github.com/bambulab/BambuStudio/issues/4669#issuecomment-2309647796

Downgrading to 555.58.02 resolved the issue for me, hopefully will be fixed in a later revision of the Nvidia drivers or Bambu Studio

jonofmac, 2025-01-20:
Here we are in 2025 with this issue still here lol. Guess we aren't considering this a bug??

eternal-sorrow, 2025-01-21:
>Guess we aren't considering this a bug??

It's not a bug, it's a feature request.

officiallymarky, 2025-01-21:
> > Guess we aren't considering this a bug??
> 
> It's not a bug, it's a feature request.

Supporting the most popular and the primary display server going forward isn't a feature request, it's necessary.

eternal-sorrow, 2025-01-21:
While I agree that native Wayland support would be nice to have, it's hardly necessary because X11 software seamlessly works on Wayland thanks to Xwayland.

officiallymarky, 2025-01-21:
> While I agree that native Wayland support would be nice to have, it's hardly necessary because X11 software seamlessly works on Wayland thanks to Xwayland.

While I'd like to agree, Bambu Slicer has issues w/ Wayland.

eternal-sorrow, 2025-01-21:
>Bambu Slicer has issues w/ Wayland

Even then, those issues could be fixed without porting the whole thing to Wayland (and probably introducing more issues along the way). In any case, this doesn't make the missing Wayland support a bug (unless the developers would consider it so).

That being said, I personally would still love to see the Wayland support added, maybe behind some kind of a switch at first for the volunteers to test and report issues back.

Alextibtab, 2025-01-31:
Just a potential temporary fix for anyone having issues, `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 WEBKIT_DISABLE_DMABUF_RENDERER=1 bambu-studio` thanks to @lkarlslund for a similar issue with orcaslicer. This lets me see the 3D preview it's a bit laggy but better than nothing at all.

officiallymarky, 2025-02-02:
> Just a potential temporary fix for anyone having issues, `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 [...]

I have really poor performance with Orca Slicer, most of the time I can get in slice and print fast enough before it is really an issue, it was really bad with Bambu Slicer, but latest versions I couldn't even get to work with Linux due to it not displaying the panels even with env vars and other tweaks.  The performance is atrocious and unusable for anything but changing a few settings and hitting slice and print.

notFunnys, 2025-02-03:
> Just a potential temporary fix for anyone having issues, `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_FORCE_COMPOSITING_MODE=1 WEBKIT_DISABLE_COMPOSITING_MODE=1 [...]

Thanks man, that works great. I replaced the exec command in the bambu-studio.desktop file with the commands. i don't even have any laggs, it's just as smooth as in the windows app

officiallymarky, 2025-02-03:
> Thanks man, that works great. I replaced the exec command in the bambu-studio.desktop file with the commands. i don't even have any laggs, it's just as smooth as in the windows app

The only one you really need is this:

`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json `

jonofmac, 2025-02-06:
> The only one you really need is this:
>
> `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json `

Can confirm. Adding only this line to the bambulabs studio launcher environmental variables worked

1) Open application menu, search for bambustudio, right click, properties
2) Application, enter environmental variables

[image]

EmilyNerdGirl, 2025-03-31:
> Can confirm. Adding only this line to the bambulabs studio launcher environmental variables worked
>
> 1. Open application menu, search for bambustudio, right click, properties
> 2. Application, enter [...]

Thank you! This works great with AppImages too on Ubuntu

snoack, 2025-08-14:
Interestingly for me setting `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json` makes things worse. If I set this env var, the home tab remains blank as well, without it only the 3D view in prepare + preview is blank.

This issue just recently started for me, Bambu Studio just worked fine a couple days ago, not exactly sure what changed, maybe some package updates. I'm using the Flatpak version.

I previously had other rendering issues in Bambu Studio on a different laptop with Nvidia graphics when using fractional scaling. But my current laptop has only Intel integrated graphics.

zish, 2025-08-29:
> 
> Give this a try
> 
> WEBKIT_DISABLE_DMABUF_RENDERER=1

This seems to have fixed things for me. My fonts seem to be rendered correctly now.

mzhaase, 2025-11-15:
I want to bring this up again since starting from Ubuntu 25.10, GNOME does not run with X11 anymore. As a result, bambu studio does not work for me on 25.10, and will also not run on the next Ubuntu LTS version.

mzhaase, 2025-11-22:
Just as a datapoint, the current release, Bambu_Studio_ubuntu-24.04_PR-8834.AppImage works on 25.10

CloudCassette, 2025-12-12:
Confirming similar performance symptoms on the Flatpak build.

## Summary
- Shortly after launch, the UI becomes very sluggish (menus, viewport interactions, general window responsiveness), consistent with the behavior described in this issue.
- I suspect Wayland may be a factor.

## Environment
- App ID: com.bambulab.BambuStudio
- Version: 2.4.0 Public Release (Flathub stable)
- Runtime: org.gnome.Platform/x86_64/48
- Flatpak: 1.16.1
- OS: Fedora Linux
- GNOME Shell: 49.2
- Session: Wayland
- Architecture: x86_64
- OpenGL: Mesa Intel(R) Graphics (RPL-U), Mesa 25.2.7 (GL 4.6 core)

## Notes
- Session is Wayland; I can retest under X11/XWayland if helpful (e.g., via QT_QPA_PLATFORM=xcb). Logs for both default (Wayland) and XCB runs are attached.
- Happy to provide longer captures or additional traces (perf/sysprof) if needed.

## Command used for X11/XWayland retest
```bash
flatpak run --env=QT_QPA_PLATFORM=xcb com.bambulab.BambuStudio
```

## Context from this thread
- Read through prior comments: several participants note Bambu Studio runs under XWayland on Wayland sessions and request native Wayland support due to glitches and performance issues.
- Reports include repeated GTK module warnings, Gtk-CRITICAL size/allocation assertions, and messages like “Cannot register URI scheme wxfs/memory more than once.” My logs show the same patterns.
- I’m attaching structured logs and exact environment details to help triage; happy to provide longer captures or profiling if useful.

## Sandbox windowing variables
Default environment observed inside the Flatpak sandbox:
```
WAYLAND_DISPLAY=wayland-0
DISPLAY=:0
GDK_BACKEND=
QT_QPA_PLATFORM=
```

## Questions for maintainers
- Should I prioritize reproductions on Wayland, XWayland (GDK_BACKEND=x11), or both?
- Is Bambu Studio expected to run natively on Wayland with wxGTK, or via XWayland? If native, which wx/GTK versions are required for support?
- Any preferred debug env vars to capture next time? Examples: G_MESSAGES_DEBUG=all, GDK_DEBUG=gl-egl, LIBGL_DEBUG=verbose, MESA_DEBUG=1, WXTRACE=all.
- Would an apitrace/sysprof capture help, and if so, which area should I focus on (windowing events vs OpenGL rendering)?
- Do you want a comparison run with the upstream AppImage to separate Flatpak packaging from application issues?
- Any GNOME settings worth toggling for repros (e.g., fractional scaling, triple buffering)?
- Should I try software rendering as a control (LIBGL_ALWAYS_SOFTWARE=1) and attach that log?
- Would xdg-desktop-portal logs be useful; if yes, which portal(s) should I enable debug for?

## Attachments
- bambu-logs-redacted.tar.gz (contains:
  - bambu-studio-default.log — default Wayland run (~60s)
  - bambu-studio-xcb.log — XCB (X11) run (~60s)
  - glxinfo.txt, flatpak-info.txt, flatpak-version.txt, permissions.txt, host-env.txt, sandbox-env.txt
  - Private/loopback IPs redacted)

[bambu-logs-redacted.tar.gz](https://github.com/user-attachments/files/24118262/bambu-logs-redacted.tar.gz)

eternal-sorrow, 2026-01-07:
Bambu studio bundles it's own patched version of WxWidgets.

dblueman, 2026-01-12:
I find frequent "BambuStudio Is Not Responding" UI modal dialogues and multi-minute hangs making release 2.4.0 unusable on Ubuntu 25.10 on x86-64.

Since everyone needing to run Ubuntu 22.04 LTS now runs Ubuntu 24.04 LTS, the former should already be removed and Bambu Lab should focus effort on Ubuntu 26.04 LTS, just weeks away to avoid user base attrition.

phisch, 2026-02-01:
I have tried to find any workaround or getting it to work as a native wayland client, but didn't find anything that improves the current situation. Bambu Studio is borderline unusable on wayland. Surfaces turn black for 3-10 seconds, the app has constant freezes for 5-infinite seconds (you gotta force quit it if it doesn't recover), it often crashes while trying to go to the device tab or sending a print.

This is a QT app, and QT can run natively on wayland. I would like to know what stops Bambu Studio from working like all other QT apps?

phisch, 2026-02-11:
@avicarpio I had to give this a test, since it sounds great, and it did seem a bit faster initially, but after it was setup, and I restarted it, it went back to being painfully slow.

Just for clarity:
- I uninstalled all instances of bambu studio and orca slicer
- downloaded the `OrcaSlicer-Linux-flatpak_PR-12119_x86_64.flatpak` from the [build you linked](https://github.com/OrcaSlicer/OrcaSlicer/actions/runs/21644455787?pr=12119)
- ran `flatpak install OrcaSlicer-Linux-flatpak_PR-12119_x86_64.flatpak`

You really had my hopes up ^^

phisch, 2026-02-12:
@avicarpio I don't think there is an AppImage in the build you linked. Am I blind?

> hey just FYI - it uses XWayland under the hood still

That's to be expected, but any performance improvement is desperately needed!

lawsonreilly, 2026-10-14:
On  Raspberry Pi Debian, the latest Bambu Studio is filled with unreadable (solid black) windows.  The "Prepare" menu mostly works, but Preview and everything to the right is unusable because I cannot read/see whatever is in the windows that appear.  This bug behaves the same whether I select light or dark desktop theme.