bambulab/BambuStudio #2599: Auto Flow rate calibration not working

Obicom, 2023-09-24:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. Go to Calibration
2. Choose Flow rate - Auto-Calibration
3. Lets the calibration run and wait for finish the job

### Actual results

Always failed -> No value 
tried it in German and English language ... both failed

### Expected results

valed value after run

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dragon-eyes-bbl, 2023-09-25:
Hi, may I know what filament did you use, and have you updated the printer firmware to [OTA version 01.06.05.01 (20230921)](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history#ota-version-01060501-20230921) or later?

Obicom, 2023-09-25:
@dragon-eyes-bbl I tried it with Giantarm PLA HS red and Eryone PLA+ green on the high temperature plate. The used firmware is the latest one (version 01.06.05.01). Tried it several times with no success.

dragon-eyes-bbl, 2023-09-25:
@Obicom Could you export the machine log and upload it to our ticket system. Once done please provide the ticket number, I can check it for you. thanks.

Obicom, 2023-09-25:
@dragon-eyes-bbl here the ticket number: TECH_Bug report_EU_US230925371001

dragon-eyes-bbl, 2023-09-26:
> @dragon-eyes-bbl here the ticket number: TECH_Bug report_EU_US230925371001

Hi @Obicom 
Thanks for the log. It looks like the filament you used is incompatible with the auto flow rate calibration.

dragon-eyes-bbl, 2023-09-26:
> @dragon-eyes-bbl here the ticket number: TECH_Bug report_EU_US230925371001

Hi @Obicom 
Thanks for the log. It looks like the filament you used is incompatible with the auto flow rate calibration. Could be the quite reflective surface finished destroying the lights. Auto flow rate requires an even surface reflection. It is possible that the filament works well with flow dynamic but fails with flow rate. 
Please try another filament with a matte finish.
Thanks.