bambulab/BambuStudio #2600: Tree support touches parts

jchu1988, 2023-09-24:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

No relevant

### Printer

X1CC

### How to reproduce

slice the uploaded file and at view layer 13 and 14.
Layer 13 is the last layer of part, layer 14 is where support overlaps with layer 13 at some area. 

### Actual results

When the part is printed, the area where support is touching the part is diffcult to remove.

[image]

### Expected results

Support should respect top and bottom Z distance. Support should also not be touching part.
Maybe introduce bottom interface layers?

### Project file & Debug log uploads

[support touching part.zip](https://github.com/bambulab/BambuStudio/files/12708875/support.touching.part.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-10-24:
Thanks for reporting the issue. We'll fix it in the next version.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan