bambulab/BambuStudio #2601: Project options not working in BambuLab Studio

p2cardon, 2023-09-24:
### Bambu Studio Version

1.7.6.9.2

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 14.0

### Additional system information

MacBook Pro M1
Apple M1 Max; , 32 Go

### Printer

BambuLab X1 Carbon + AMS

### How to reproduce

1/ Lauch bambu Studio
2/ Create a new project
3/ Click "project" pane

None of the options works : define creator name, model description, accessories, profile name, profile description …

Project advantages and workflow are described in your WIKI, unfortunately this doesn't works

### Actual results

Software feature not working at all

### Expected results

Should work at least as described in the WIKI

### Project file & Debug log uploads

[Uploading log.zip…]()
[Uploading sample.3mf.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

MadMax389, 2023-09-24:
I'm still waiting for this feature to reappear since it disappeared in 1.7.5

walterwongbbl, 2023-09-25:
@p2cardon @MadMax389 
The project can only open model files downloaded from the marketplace.
Private model files are not supported. Please visit the following link.
https://makerworld.com/

p2cardon, 2023-09-25:
Can't be satisfied of such a "non answer". User Manual which is inside bambulab Studio as a dedicated chapter dedicated to "Project based workflow". It links to this page : https://wiki.bambulab.com/en/software/bambu-studio/project-based-workflow … now where it's specified this dedicated to Market place … If so, I think the online help should be modified and that the "Project" pane should be hidden if not appropriated … Many user say this was working in previous versions …

walterwongbbl, 2023-10-13:
Hi.  @p2cardon 
so sorry.  Our colleagues responsible for the wiki didn't close this wiki page in a timely manner, leading to some misunderstandings. We will close this window and add additional explanations.

closed 2023-12-21 by walterwongbbl