bambulab/BambuStudio #2602: Project Information

mast85-fm, 2023-09-24:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

1. Create New Project
2. Save New Project
3. Go to "Project Tab"
4. Try to Write some project information

### Actual results

Is no possible write project info
nothing happends if i clic on : Project Info, Project Accessories, Project Profile Informations

### Expected results

I expect that the entire Project tab works good

### Project file & Debug log uploads

[Untitled.zip](https://github.com/bambulab/BambuStudio/files/12709569/Untitled.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-09-25:
#2601

walterwongbbl, 2023-09-25:
Hi. @mast85-fm 
The project can only open model files downloaded from the marketplace.
Private model files are not supported. Please visit the following link.
https://makerworld.com/

mast85-fm, 2023-09-25:
So in the new version of BambuStudio I can’t create my own project with
informations and photos as was at the beginning?! Can I ask you why this
choise?! In the past I’ve created many project ad I think was a really good
and helpful feature for BambuStudio.

walterwongbbl, 2023-09-25:
@mast85-fm 
I apologize for the confusion. We have not removed these features. We have relocated the capability to add information and photos to our model marketplace. You can upload your files and add information and photos in the model marketplace.

closed 2023-10-13 by walterwongbbl