bambulab/BambuStudio #2603: Black color is accurate but not usefully rendered in Bambu Studio

sprior, 2023-09-25:
[image]

**Is your feature request related to a problem? Please describe.**
While I appreciate that when I configure a black filament in Bambu Studio that you'd like to render it as accurately as possible it comes out almost featureless on the screen and very difficult to see or work with.  I know I can configure any color I want in Bambu Studio itself, but I think a very common usecase is to load filament in the AMS, select black on the printer then sync those colors to Bambu Studio and the screenshot shows how that comes out.

**Describe the solution you'd like**
I think rendering black as a dark grey might be easier to actually work with.

**Describe alternatives you've considered**
Maybe a config switch to adjust colors to show detail (maybe call it HDR mode?) might be a good option.

QingZhangBambu, 2023-09-25:
we will fix it soon.

Haidiye00, 2023-09-25:
Thank you. We have fixed this issue in the upcoming V1.8 version. Stay tuned

Haidiye00, 2024-01-16:
The issue has been fixed, please use the latest version V1.8

closed 2024-01-16 by Haidiye00