bambulab/BambuStudio #2605: Linux: Clicking edit filament symbol segvaults/crashes

norpol, 2023-09-25:
### Bambu Studio Version

01.07.06.92 (and 1.6.6)

### Where is the application from?

Built from source by yourself or others

### OS version

NixOS 23.05

### Additional system information

AMD Ryzen, swaywm, NixOS

### Printer

Bambulab X1

### How to reproduce

1. Open BambuStudio
2. Go to Prepare
3. Click Edit Symbol next to the filament name
4. `segmentation fault (core dumped)` and window is closed

Issue reported and now also verified for me. Patch that workarounds the issue was shared in ongoing attempt to get BambuStudio packaged in NixOS https://github.com/NixOS/nixpkgs/pull/206495#issuecomment-1728971710 : 

Link to the patch that is working with `01.07.06.92` for me
https://github.com/ovlach/nix-orca-slicer/blob/1e71e6e01ef3b49d0731ca611b6bfce32dbe26d0/0001-segv-patches.patch

### Actual results

Crash

### Expected results

No crash

### Project file & Debug log uploads

Logs don't show the crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-10-24:
hello, @norpol 

I am sorry we missed this issue before
we have tried different versions on ubuntu and fedora
and can not reproduce this issue

do you still have this issue?
if yes,  Could you help to do more do dig into it?

what is the image size after` wxImage image(btn->GetBitmap().GetSize());`
what is the size of the bitmap?

lanewei120, 2023-10-30:
> Hey, no worries @lanewei120. Do you have any suggestion how I could best collect that information for you?

@norpol 

just the image size and detailed callstack will be ok

and also is it a 100% reproducible issue?

norpol, 2023-12-03:
Hey @lanewei120, I apologize for my delayed reply. I've had some trouble re-compiling bambu-studio but resolved it last weekend. As of writing, with the most recent beta release 01.18... the error is still reproducible.

I can offer you the gdb callstack, I tried adding print statements to the file in question, but didn't succeed and my roundtrip of re-compiling is like 20 Minutes in worst case, which is making it a bit hard for me to provide more. 

I've also tried orca-slicer @1.8.0 beta and the error doesn't happen there (exact same compilation environment).

```bash
$ echo -e "run\nbt\n\n" | LC_ALL=C gdb -- bambu-studio
[...]
(gdb) #0  0x00007ffff073b9e3 in wxImage::InitAlpha() ()
   from /nix/store/hnlrhsxic6s9076dl5v07kby00gdc378-wxwidgets-3.1.7/lib/libwx_gtk3u_core-3.1.so.7
#1  0x0000000001425e22 in Slic3r::GUI::ColourPicker::set_undef_value(wxColourPickerCtrl*) ()
#2  0x000000000142622b in Slic3r::GUI::ColourPicker::set_value(boost::any const&, bool) ()
#3  0x000000000142cbee in Slic3r::GUI::ConfigOptionsGroup::reload_config() ()
#4  0x000000000137897b in Slic3r::GUI::Page::activate(Slic3r::ConfigOptionMode, std::function<void ()>) ()
#5  0x000000000138241b in Slic3r::GUI::Tab::activate_selected_page(std::function<void ()>) ()
#6  0x000000000137848e in Slic3r::GUI::Tab::OnActivate() ()
#7  0x000000000138e1aa in Slic3r::GUI::Tab::tree_sel_change_delayed(wxCommandEvent&) ()
#8  0x000000000138e333 in wxEventFunctorFunctor<wxEventTypeTag<wxCommandEvent>, Slic3r::GUI::Tab::create_preset_tab()::{lambda(wxCommandEvent&)#11}>::operator()(wxEvtHandler*, wxEvent&) ()
#9  0x00007ffff0c4dc52 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /nix/store/hnlrhsxic6s9076dl5v07kby00gdc378-wxwidgets-3.1.7/lib/libwx_baseu-3.1.so.7
#10 0x00007ffff0c4df87 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
```

Full gdb trace
[gdb.txt](https://github.com/bambulab/BambuStudio/files/13539560/gdb.txt)

norpol, 2023-12-03:
@lanewei120 I found the PR that addresses the bug in OrcaSlicer, though it's more of a safeguard and to me unclear what introduced the underlying issue in the first place.

https://github.com/SoftFever/OrcaSlicer/pull/2220/files

aacebedo, 2023-12-10:
I have the same issue on NixOS and wayland. @lanewei120 Did you reproduce it ?

norpol, 2023-12-10:
@aacebedo https://github.com/bambulab/BambuStudio/releases/tag/v01.08.01.57 check the release notes (and this issue) should be resolved now.

norpol, 2023-12-10:
https://play.google.com/store/apps/details?id=com.streetwriters.notesnook

norpol, 2023-12-10:
Verified to be fixed with latest beta, thanks

https://github.com/NixOS/nixpkgs/pull/273350#issuecomment-1848969734

closed 2023-12-10 by norpol