bambulab/BambuStudio #2606: OpenVDB / OpenEXR 2.x + IlmBase -> OpenEXR 3.x + Imath

cederom, 2023-09-25:
### Bambu Studio Version

git/master

### Where is the application from?

Bambu Lab github releases, Built from source by yourself or others

### OS version

FreeBSD RELEASE 13.2 AMD64

### Printer

X1C

### How to reproduce

1. CMakeLists.txt contains `OpenVDB` reference to obsoleted `IlmBase`. Part of `IlmBase` became `OpenEXR` version 3 + `Imath` library [1].

https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/CMakeLists.txt#L549
https://github.com/bambulab/BambuStudio/blob/5ef759ce41863f989da7b363f7c94e5edc5ade0d/CMakeLists.txt#L551

3. Modern `PrusaSlicer` already uses new `OpenEXR` + `Imath`.

[1] https://openexr.com/en/latest/PortingGuide.html?highlight=ilmbase#background

### Actual results

* I am porting BambuStudio to FreeBSD OS :-)
* There was `IlmBase` port in FreeBSD but due to `OpenEXR` updates the port is now gone and code went to `OpenEXR` and `Imath` following the upstream.
* FreeBSD port of PrusaSlicer already uses `OpenEXR` + `Imath`.

### Expected results

Dependencies from `IlmBase` are updated to `OpenEXR` + `Imath` :-)

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

harley-codes, 2023-10-06:
Good find @cederom - was getting stuck on this issue too. Hopefully this will get resolved soon.

lanewei120, 2023-10-07:
@cederom 

thanks for this suggestion
"Modern PrusaSlicer already uses new OpenEXR + Imath."

could you show us how did you find this?
I just checked the OpenVDB dependence, and find no difference on this logic

cederom, 2023-10-07:
Thank you @harley-codes and @lanewei120 for your interest in the problem :-)

Here are three main resources to reference:
1. https://cgit.freebsd.org/ports/tree/misc/openvdb
2. https://cgit.freebsd.org/ports/tree/cad/PrusaSlicer
3. https://cgit.freebsd.org/ports/tree/graphics/openexr
4. https://openexr.com/en/latest/PortingGuide.html?highlight=ilmbase
5. https://cgit.freebsd.org/ports/tree/math/Imath

1 is the source of FreeBSD's port for OpenVDB. If you look at Makefile line 38 it states OpenVDB depends on OpenEXR (here: https://cgit.freebsd.org/ports/tree/misc/openvdb/Makefile).

2 is the source of FreeBSD's port for PrusaSlicer version 2.6.0. If you look at Makefile line 21 it states libImath dependency (here: https://cgit.freebsd.org/ports/tree/cad/PrusaSlicer/Makefile).

3 is the source of FreeBSD's port for OpenEXR. If you look at Makefile line 2 it states version used is 3.2.1 (here: https://cgit.freebsd.org/ports/tree/graphics/openexr/Makefile).

4 is the porting guide of OpenEXR that states changes between version 2 and version 3, particularly the "_OpenEXR and Imath Are Different Packages_" section states "**The parts of “IlmBase” that were Imath and half are now repackaged as the Imath library. The IlmThread and Iex libraries have been folded into the OpenEXR package, since they were were not necessary to the rest of Imath.**"

5 is the dedicated FreeBSD port for Imath following the changes in OpenEXR version 3.X. That is a dependency for PrusaSlicer version 2.6.0 (2). It replaces the old version 2 ilmbase/OpenEXR with version 3 (see commit https://cgit.freebsd.org/ports/commit/?id=05044ea01992c28028cc87f374560aad1b6c595d).

I hope that helps :-) Please let me know how can I help further :-)

lanewei120, 2023-10-08:
> Thank you @harley-codes and @lanewei120 for your interest in the problem :-)
> 
> Here are several resources to reference:
> 
> 1. https://cgit.freebsd.org/ports/tree/misc/openvdb
> 2. https://cgit.freebsd.org/ports/tree/cad/PrusaSlicer
> 3. https://cgit.freebsd.org/ports/tree/graphics/openexr
> 4. https://openexr.com/en/latest/PortingGuide.html?highlight=ilmbase
> 5. https://cgit.freebsd.org/ports/tree/math/Imath
> 
> 1 is the source of FreeBSD's port for OpenVDB. If you look at Makefile line 38 it states OpenVDB depends on OpenEXR (here: https://cgit.freebsd.org/ports/tree/misc/openvdb/Makefile).
> 
> 2 is the source of FreeBSD's port for PrusaSlicer version 2.6.0. If you look at Makefile line 21 it states libImath dependency (here: https://cgit.freebsd.org/ports/tree/cad/PrusaSlicer/Makefile).
> 
> 3 is the source of FreeBSD's port for OpenEXR. If you look at Makefile line 2 it states version used is 3.2.1 (here: https://cgit.freebsd.org/ports/tree/graphics/openexr/Makefile).
> 
> 4 is the porting guide of OpenEXR that states changes between version 2 and version 3, particularly the "_OpenEXR and Imath Are Different Packages_" section states "**The parts of “IlmBase” that were Imath and half are now repackaged as the Imath library. The IlmThread and Iex libraries have been folded into the OpenEXR package, since they were were not necessary to the rest of Imath.**"
> 
> 5 is the dedicated FreeBSD port for Imath following the changes in OpenEXR version 3.X. That is a dependency for PrusaSlicer version 2.6.0 (2). It replaces the old version 2 ilmbase/OpenEXR with version 3 (see commit https://cgit.freebsd.org/ports/commit/?id=05044ea01992c28028cc87f374560aad1b6c595d).
> 
> Switching BambuStudio / PrsaSlicer to OpenEXR 3.X + Imath would make porting to new platforms possible, also should improve maintenance on existing platforms :-)
> 
> I hope that helps :-) Please let me know how can I help further :-)

thanks for the detailed information

I just found there is no modifications in https://github.com/prusa3d/PrusaSlicer to support OpenEXR+Imath
do you mean PrusaSlicer can be compiled under FreeBSD without the modifications in its original git, 
and just do the modifications in https://cgit.freebsd.org/ports/tree/cad/PrusaSlicer?

cederom, 2023-10-09:
* Okay this is problem of OpenVDB that still depends on OpenEXR 2.X + IlmBase not the OpenEXR 3.X (3.2.1 is out already!) and Imath :-(
* I can see that FreeBSD port patches locally the OpenVDB 10.0.1 before build in order to use Imath in place of IlmBase. I will try to send those patches to OpenVDB upstream that should solve our problems :-)
* Please bump the https://github.com/AcademySoftwareFoundation/openvdb/issues/1541 so they know this is important issue :-)

cederom, 2023-10-09:
I will try to provide patches to both OpenVDB and PrusaSlicer:
1. Transition to OpenEXR 3.X + Imath.
2. Build out of the box on FreeBSD (I guess this will be also mandatory to build BambuStudio here).

cederom, 2023-10-09:
@lanewei120 is it possible to skip PrusaSlicer build as part of BambuStudio and use existing installed system package or is it hardcoded part?

That would allow easy building of BambuStudio that could depend on the already installed system packages (no matter how these were built).

If this is not possible then we need to patch the upstream of OpenVDB and PrusaSlicer in the first place and then maintain the FreeBSD support in the upstream.

lanewei120, 2023-10-09:
> @lanewei120 is it possible to skip PrusaSlicer build as part of BambuStudio and use existing installed system package or is it hardcoded part?
> 
> That would allow easy building of BambuStudio that could depend on the already installed system packages (no matter how these were built).
> 
> If this is not possible then we need to patch the upstream of OpenVDB and PrusaSlicer in the first place and then maintain the FreeBSD support in the upstream.
> 
> Update: I can see that BambuStudio IS a modified PrusaSlicer. There was no question ;-)

yes, the building of BambuStudio is dependent of PrusaSlicer
as BambuStudio is a based on PrusaSlicer, the building steps are very similiar
thanks for doing this~

cederom, 2023-10-10:
@lanewei120 is BambuStudio 1:1 copy of the PrusaSlicer + additional "out-of-tree" modifications, so patching PrusaSlicer upstream will also propagate BambuStudio or these projects diverged at some point and need to be patched separately and will not stay in sync?

lanewei120, 2023-10-10:
> @lanewei120 is BambuStudio 1:1 copy of the PrusaSlicer + additional "out-of-tree" modifications, so patching PrusaSlicer upstream will also propagate BambuStudio or these projects diverged at some point and need to be patched separately and will not stay in sync?
> 
> I mean the best possible situation would be to patch the PrusaSlicer and then rebase changes to BambuStudio.. now.. and in future :-)

need to patch seperately:)
they are seperate code base now

cederom, 2023-10-10:
Okay @lanewei120, so the BambuStudio was created out of PrusaSlicer, do you share back fixes and updates to the upstream PrusaSlicer? o_O

lanewei120, 2023-10-10:
> Okay @lanewei120, so the BambuStudio was created out of PrusaSlicer, do you share back fixes and updates to the upstream PrusaSlicer? o_O

they have much difference in lots of parts,
and the PrusaSlicer's maintainer will do this if they think it is neccessary

cederom, 2023-12-29:
Okay, OpenVDB 11 was released with required fixes in the upstream https://github.com/AcademySoftwareFoundation/openvdb/pull/1678 :-)

The other stuff is up to PrusaSlicer upstream. I know more now how BambuStudio is organized. I will get back to porting in my free time :-)

Thank you for the support and time! :-)

closed 2023-12-29 by cederom