bambulab/BambuStudio #2607: Engineering Plate won't slice with PETG part with PLA support interface

bCaff, 2023-09-25:
### Bambu Studio Version

1.4.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

Bambu Studio will not slice a part with material selected as PETG but that has support interface material selected as PLA when applied with the Engineering Plate.  I receive an error message as shown in the screenshot attached.

I think this has to do with the 'Prepare' simulation of the purge block that allows the PLA material slot 4 to touch the engineering plate.  I don't recall this issue before the latest update, but maybe I haven't printed this configuration on the engineering plate before.  

If the file would slice, then there shouldn't be any PLA touching the actual be
d, making the engineering plate sufficient for printing PETG parts to start.

[image]

### Actual results

Bambu Studio won't allow me to slice the file at all even though the error technically isn't true.

### Expected results

Bambu Studio should slice and print this file since no PLA will actually touch the bed.

### Project file & Debug log uploads

[Bambu Jet Engine Casings.zip](https://github.com/bambulab/BambuStudio/files/12717318/Bambu.Jet.Engine.Casings.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

Xiuqi-Huang-CN, 2023-09-26:
Because the Engineering Plate is not compatible with PLA, and the reason for this is that the Engineering Plate needs high enough temperatures to provide strong enough first layer bonding forces to enable the models to be printed and finished. And the temperature for PETG is 65 to 75 ℃, and the temperature for PLA is 70 to 85 ℃ which is too high for PLA: clogging and damage can happen at this high temperature.
It's recommended to use a High Tempt Plate or a PEI Plate to print PETG and PLA at the same time (not mixed filaments for models, but PLA or PETG just serve as the interface of the support). And if you only has one Engineering Plate, you can choose the High Tempt Plate or PEI Plate to slice, and apply glue to the Engineering Plate to provide strong enough first layer bonding forces to both PETG and PLA and pretend that it is an High Tempt Plate or PEI Plate (not really recommended, or the leveling effect may be different).
Hope this may help you.

XunZhangBambu, 2024-08-09:
@bCaff  Won't do. Only after the slicing is completed can we know the material used in the first layer, which needs to be tested with the bed type, causing many unnecessary slices.

closed 2024-08-09 by XunZhangBambu

ostryhub, 2024-12-03:
I have the same problem, the PETG is the only filament touching the bed plate where PLA is the support interface only.

wuast94, 2025-05-22:
same, an option to say i know what i do would be nice