bambulab/BambuStudio #2608: Detect user installing plate after bed leveling. Perform final bed height check before putting down filament.

UnglazedSquash, 2023-09-25:
**Is your feature request related to a problem? Please describe.**
I have a cousin who got a X1C and he is finding all kinds of new ways to mess up. I think there are some valuable lessons being
learned from my helping him. 

Perform a final bed height check right before laying down the filament. My cousin forgot to put the plate on, when the printer started printing it stopped and threw the no plate detected do you want to continue? My cousin placed the plate onto the bed and hit continue. The print head was then way too low to the engineering plate + cool plate. By checking the bed height right before printing you could detect if the user has installed the plate after the bed leveling was done. Then flag and perform the auto bed leveling again.

**Describe the solution you'd like**
Right before laying down filament double check the bed height. This will detect if the user has installed the plate after auto bed leveling and prevent users from doing this and having failed prints because the auto bed leveling happened with no plate and now there is a plate there.

**Describe alternatives you've considered**
Some logic to detect during flow calibration or the back right corned line if moving the tool head is requiring more torque than expected because the nozzle is into the bed material due to the bed height being off.

**Additional context**

SaltWei, 2023-09-26:
The bed level data is useless after installing plate. The touch surface is different and the bed is also curving.

UnglazedSquash, 2023-09-26:
@SaltWei What it seemed like was:
1. User starts print without plate on bed
2. User realizes plate is not there because printer pauses print with warning
3. User installs plate onto bed
4. User acknowledges popup on printer for "plate not detected (did not see code) do you want to continue" and says to continue
5. Print starts and the nozzle height is too low because it probed the bed when there was no plate there
6. print continues with no filament laying down because nozzle is jammed into bed. 
7. cool plate surface gets damaged from nozzle to far down

SaltWei, 2023-09-26:
I see. I think the printer pop-up windows is mess which lead to the damage of bed. @sodachen22 Please check this issue and try to improve the sequence and pop-up message.