bambulab/BambuStudio #2611: "Send print job to" window wont close

Anmin27, 2023-09-25:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

CPU ryzen 7950x
RAM 32g 6000mh
GPU Nvidia RTX 4090

### Printer

x1 carbon

### How to reproduce

After clicking Print plate, the "Send print job to" window opens as expected, i usually only have bed level and enable ams selected. i click "Send". the Print start and Bambu Studio switches to Device tab as expected but from then on the "Send print job to" window will not close unless I close Bambu studio completely. 

### Actual results

[image]

### Expected results

The "Send print job to" window should auto close once the print job has been sent to printer and Bambu Studio has changed to Device tab.

### Project file & Debug log uploads

[SendprintjobtoBUG.zip](https://github.com/bambulab/BambuStudio/files/12719943/SendprintjobtoBUG.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12719944/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-09-26:
Hi. @Anmin27 
Is this the same with the previous version

Anmin27, 2023-09-27:
Hey, @walterwongbbl 
Yeah, this has been an issue for me for the current and previous 2 versions I believe. It's gotten to where I'm used to just moving that specific window out of the way to the side as it doesn't interfere with any thing in Bambu Studio and even refreshes when sending a new print job, I just have to move it out of the way each time.

walterwongbbl, 2023-10-13:
Hi. @Anmin27 
Sorry, we cannot reproduce this issue.  
Can you record a video? We would like to confirm how you operated it. Under normal circumstances, this dialog box will be closed after sending and printing.

Anmin27, 2023-10-13:
Oddly enough this only happened a few more times after sharing. The issue has since resolved itself. I wish I knew how it resolved itself or why it was happening to begin with. I will be sure to try to capture it if it ever happens again.     Thank you so much for looking into the problem with such haste and effort.

walterwongbbl, 2023-10-16:
Hi. @Anmin27 
I'm very sorry, if you encounter this issue again next time, please contact us again.
The current issue will be closed.

closed 2023-10-16 by walterwongbbl