bambulab/BambuStudio #2612: in regard of by object print sequence...

dukuby, 2023-09-26:
I desperately want that bambu studio let users determine the presence of a collision by their responsibility.
Even though the object's height is higher than the gantry limit, if objects placed very carefully in order, we can avoid the collision and could get a wonderful result.
Please check my video below;
https://youtu.be/vKZju3_AM9Y?si=jfSmTC2j9ZeEgfcU

I strongly want to do this kind of multi-parts of different colors in one plate printing.   
Very thanks!

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-10-07:
We can add one option in preferences. If user enable the option, the slicer will not limit the space and height of objects when printing by object sequence.
[image]

Can this be acceptable?

leuys, 2025-07-09:
This would be greatly appreciated! Even just as a quick work-around for now, once a better print-by-object collision detection is implemented.

Q-V-91, 2026-02-04:
I Think the last update got too strict, i used to print fine 3 models by object and now it doesn't let me do it because of height, it would be even better if we could choose the order of the part printed by object. 

Never had this issue before the big 2.5 update with this exact file. Printed fine on A1 and P1s

[image]