bambulab/BambuStudio #2613: No Resume after 2nd Pause in Print

Who-The-Hell-am-I, 2023-09-26:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Printer

BambuLab X! Carbon with 2 AMS

### How to reproduce

Add a Pause before Line 1
Add another Pause before Line 2
Slice Plate

### Actual results

Printer Pauses after Calibration, allowing me to remove calibration pattern and re-glue the area, Resume is available and functions normally.  Printer then pauses after first layer is complete, allowing me to inspect the first layer before printing a 10+ hour model.  This time there is no Resume button, only Pause, Stop and Close.  Pause does nothing, Stop will begin the Cancel process.  I go Back after pressing Stop and getting the warning.  
I am able to continue the print by powering off the printer with the switch on the back, then turning it back on.  This allows the power failure recovery to give me the option to resume.
This has happened 4 times in the last 2 days - at first with the last version of Bambu Studio, I updated today, but the same problem exists.
[image]

### Expected results

I should get a resume option after the 2nd Pause.
[image]

### Project file & Debug log uploads

[Utensils.gcode.zip](https://github.com/bambulab/BambuStudio/files/12721048/Utensils.gcode.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

keyoss, 2023-11-09:
refs #2860

keyoss, 2023-11-09:
refs #1389