bambulab/BambuStudio #2614: tree manual support problem

riccalisci, 2023-09-26:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

in manual tree support , slicing is incorrect.

### Actual results

there aren't no supports where it should be.
[image]

### Expected results

correct support

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12724560/log.zip)
[block.zip](https://github.com/bambulab/BambuStudio/files/12724561/block.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2023-10-24:
Thanks for reporting the issue. We'll fix it in the next version.

tangketan, 2023-12-21:
Offical 1.8 has been released. Close this issue.

closed 2023-12-21 by tangketan