bambulab/BambuStudio #2616: Colour Blind Friendly

k0158393, 2023-09-26:
I am very colour blind, so I struggle to identify individual colours when next to a group of others such as a colour wheel or colour board. It is therefore hard when trying to set colours when using the AMS and keeping track of what is loaded.

It would be beneficial for me and I imagine others if there was an option to tag up the colours aswell either with the full word or an abbreviated, e.g. Red or R.

This wouldn't need to be forced upon everyone, it could just be an option which is selected, similar to when colour blind people play certain video games.

norpol, 2023-10-03:
Looking in the forum also shows other people requesting that, even with similar proposal mock graphics. Are there other places where this is an issue?

* https://forum.bambulab.com/t/user-friendly-filament-names-colorblind/9452
* https://forum.bambulab.com/t/labelling-filament-colors/12822

UnglazedSquash, 2023-10-14:
@norpol  Hey! Here is a great article that shows visually how to improve color-blind accessibility. It is something like 5-10% of the population has some form of colorblindness. The added benefit of developing for colorblindness is it also improves the usability for non-colorblind users by making things even more obvious.

[https://www.secretstache.com/blog/designing-for-color-blind-users/](https://www.secretstache.com/blog/designing-for-color-blind-users/)