bambulab/BambuStudio #2623: P1S Calibration Problem

Rackl-p, 2023-09-27:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Prozessor	Intel(R) Core(TM) i9-9900KF CPU @ 3.60GHz   3.60 GHz
Installierter RAM	64,0 GB
Geräte-ID	BCDF6B3E-5CE8-4BFD-9E69-C268A915FA02
Produkt-ID	00330-80216-60522-AA613
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor
Stift- und Toucheingabe	Für diese Anzeige ist keine Stift- oder Toucheingabe verfügbar.

### Printer

Bambu Lab P1S

### How to reproduce

Flow Calibration gives Network Problems, 
it takes a long time to start and stops all the time around 65%. 

### Actual results

Stops at the second or third line

### Expected results

Run trough

### Project file & Debug log uploads

o

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-10-07:
Hi. @Rackl-p 
Does this issue occur when sending a print normally? If it does, please check the status of your computer network and the printer network, and try improving the signal by adjusting the printer's placement.

Rackl-p, 2023-10-07:
Hi, If i send a normal file, everything works fine.My friend has the some problem at his home, normal prints working very fine, but the calibration stops and don’t run trough the complete cycle. Mit freundlichen Gruessen / Best regardsPatrick RacklCAM Spezialistsolid AI UGAm Weinberg 1035469 AllendorfGermany Mobil +49 151 19192199Fax +49 6407 ***@***.*** AI UGGeschaeftsfuehrung/Board of Management: Matthias KramolischGesellschaft/Registered Office: Allendorf (Lumda)Handelsregister/Commercial Register: HRB10476 (Amtsgericht Gießen)___________________________________________________________________________________________Bitte beachten Sie: Diese E-Mail kann vertrauliche und/oder rechtlich geschützte Informationen enthalten.Der Inhalt ist ausschließlich für den bezeichneten Adressaten bestimmt.Jede Form der Veröffentlichung, Vervielfältigung oder Weitergabe des Inhalts fehlgeleiteter E-Mails ist unzulässig.Please note that this e-mail may contain confidential and/or legally protected information.The content is exclusively destined for the named addresses.Any form of publication, distribution or forwarding of the content of misdirected e-mails is illegitimate.___________________________________________________________________________________________

zhimin-zeng-bambulab, 2023-10-11:
@Rackl-p Could you show me a picture of this bug?

zhimin-zeng-bambulab, 2023-10-13:
@Rackl-p  Could you describe your problem in more detail?

zhimin-zeng-bambulab, 2024-02-05:
@Rackl-p  Does this problem still exist in the latest version? If so, could you share the log files?

closed 2024-06-26 by zhimin-zeng-bambulab