bambulab/BambuStudio #2624: Docker image does not build

xaionaro, 2023-09-27:
### Bambu Studio Version

5ef759ce41863f989da7b363f7c94e5edc5ade0d

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 23.04

### Printer

Bambu X1C

### How to reproduce

cd /tmp
git clone https://github.com/bambulab/BambuStudio
cd BambuStudio
./DockerBuild.sh

### Actual results

/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `Slic3r::sla::generate_interior(Slic3r::TriangleMesh const&, Slic3r::sla::HollowingConfig const&, Slic3r::sla::JobController const&)':
Hollowing.cpp:(.text+0x2924): undefined reference to `Slic3r::mesh_to_grid(indexed_triangle_set const&, openvdb::v8_2::math::Transform const&, float, float, float, int)'
/usr/bin/ld: Hollowing.cpp:(.text+0x2b05): undefined reference to `Slic3r::redistance_grid(openvdb::v8_2::Grid<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::LeafNode<float, 3u>, 4u>, 5u> > > > const&, double, double, double)'
/usr/bin/ld: Hollowing.cpp:(.text+0x2c13): undefined reference to `Slic3r::grid_to_mesh(openvdb::v8_2::Grid<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::LeafNode<float, 3u>, 4u>, 5u> > > > const&, double, double, bool)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `openvdb::v8_2::math::ScaleTranslateMap::postRotate(double, openvdb::v8_2::math::Axis) const':
Hollowing.cpp:(.text._ZNK7openvdb4v8_24math17ScaleTranslateMap10postRotateEdNS1_4AxisE[_ZNK7openvdb4v8_24math17ScaleTranslateMap10postRotateEdNS1_4AxisE]+0x7e): undefined reference to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `openvdb::v8_2::math::ScaleTranslateMap::preShear(double, openvdb::v8_2::math::Axis, openvdb::v8_2::math::Axis) const':
Hollowing.cpp:(.text._ZNK7openvdb4v8_24math17ScaleTranslateMap8preShearEdNS1_4AxisES3_[_ZNK7openvdb4v8_24math17ScaleTranslateMap8preShearEdNS1_4AxisES3_]+0xcd): undefined reference to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `openvdb::v8_2::math::ScaleTranslateMap::preRotate(double, openvdb::v8_2::math::Axis) const':
Hollowing.cpp:(.text._ZNK7openvdb4v8_24math17ScaleTranslateMap9preRotateEdNS1_4AxisE[_ZNK7openvdb4v8_24math17ScaleTranslateMap9preRotateEdNS1_4AxisE]+0x7e): undefined reference to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `openvdb::v8_2::math::ScaleTranslateMap::postShear(double, openvdb::v8_2::math::Axis, openvdb::v8_2::math::Axis) const':
Hollowing.cpp:(.text._ZNK7openvdb4v8_24math17ScaleTranslateMap9postShearEdNS1_4AxisES3_[_ZNK7openvdb4v8_24math17ScaleTranslateMap9postShearEdNS1_4AxisES3_]+0xc3): undefined reference to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `openvdb::v8_2::math::ScaleMap::postRotate(double, openvdb::v8_2::math::Axis) const':
Hollowing.cpp:(.text._ZNK7openvdb4v8_24math8ScaleMap10postRotateEdNS1_4AxisE[_ZNK7openvdb4v8_24math8ScaleMap10postRotateEdNS1_4AxisE]+0x191): undefined reference to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)'
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o):Hollowing.cpp:(.text._ZNK7openvdb4v8_24math8ScaleMap9preRotateEdNS1_4AxisE[_ZNK7openvdb4v8_24math8ScaleMap9preRotateEdNS1_4AxisE]+0x191): more undefined references to `openvdb::v8_2::math::simplify(std::shared_ptr<openvdb::v8_2::math::AffineMap>)' follow
/usr/bin/ld: libslic3r/liblibslic3r.a(Hollowing.cpp.o): in function `void openvdb::v8_2::io::readCompressedValues<float, openvdb::v8_2::util::NodeMask<3u> >(std::istream&, float*, unsigned int, openvdb::v8_2::util::NodeMask<3u> const&, bool)':
Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x4a): undefined reference to `openvdb::v8_2::io::getStreamMetadataPtr(std::ios_base&)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x59): undefined reference to `openvdb::v8_2::io::getDataCompression(std::ios_base&)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x99): undefined reference to `openvdb::v8_2::io::getFormatVersion(std::ios_base&)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0xd7): undefined reference to `openvdb::v8_2::io::getGridBackgroundValuePtr(std::ios_base&)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x223): undefined reference to `openvdb::v8_2::math::internal::half::_toFloat'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x388): undefined reference to `openvdb::v8_2::io::getFormatVersion(std::ios_base&)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x6ff): undefined reference to `openvdb::v8_2::io::StreamMetadata::delayedLoadMeta() const'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x715): undefined reference to `openvdb::v8_2::io::StreamMetadata::gridMetadata()'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x82b): undefined reference to `openvdb::v8_2::io::DelayedLoadMetadata::getMask(unsigned long) const'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x83f): undefined reference to `openvdb::v8_2::io::unzipFromStream(std::istream&, char*, unsigned long)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x8a6): undefined reference to `openvdb::v8_2::io::bloscFromStream(std::istream&, char*, unsigned long)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x8b8): undefined reference to `openvdb::v8_2::io::DelayedLoadMetadata::getCompressedSize(unsigned long) const'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0x8d8): undefined reference to `openvdb::v8_2::io::bloscFromStream(std::istream&, char*, unsigned long)'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0xa0c): undefined reference to `openvdb::v8_2::io::StreamMetadata::leaf() const'
/usr/bin/ld: Hollowing.cpp:(.text._ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b[_ZN7openvdb4v8_22io20readCompressedValuesIfNS0_4util8NodeMaskILj3EEEEEvRSiPT_jRKT0_b]+0xa23): undefined reference to `openvdb::v8_2::io::unzipFromStream(std::is [... cut ...]
make[3]: *** [src/CMakeFiles/BambuStudio.dir/build.make:185: src/bambu-studio] Error 1
make[2]: *** [CMakeFiles/Makefile2:596: src/CMakeFiles/BambuStudio.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:603: src/CMakeFiles/BambuStudio.dir/rule] Error 2
make: *** [Makefile:236: BambuStudio] Error 2

### Expected results

Successful build of a Docker image.

### Project file & Debug log uploads

The studio does not compile, thus I cannot run it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

xaionaro, 2023-10-17:
> do you still have this issue?

I don't know. I already decided not to buy Bambu (given I wasn't able to even just build&run the Slicer) :)
Feel free to close the issue.

lanewei120, 2023-10-17:
Have a nice day:)

closed 2023-10-17 by lanewei120