bambulab/BambuStudio #2625: Auto refill

Daggers62, 2023-09-27:
I have two rolls of Bambu Labs PLA Matte, one roll is red and has around 10 metres left, the other roll is blue and is almost full. I want to use the red first and then use the blue. When selecting Auto Refill it will not allow me to do this because one roll is red and the other is blue. 

There should be an on screen prompt to allow the user to choose if this is what they require.

nihilisticzebra, 2023-10-03:
Yeah, that's annoying I know with 3rd party filaments I just change the color to match

Merlwynd, 2023-10-03:
You probably know this, but as workaround, pretty simple to splice your red filament to the blue and roll remaining red filament onto blue roll and you’re set to go…