bambulab/BambuStudio #2626: Font issues info screen unreadable

khmewes, 2023-09-27:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Version	10.0.19045 Build 19045

### Additional system information

Core i5 gen 4 16GB RAM

### Printer

Creality Ender 3S1

### How to reproduce

Font unreadable see screenshot 
[image]
picture

### Actual results

[image]

### Expected results

Expect the font to be readable

### Project file & Debug log uploads

[Lid5_short.zip](https://github.com/bambulab/BambuStudio/files/12739959/Lid5_short.zip)
[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/12739961/BambuStudio.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

LiZ-Li-BBL, 2023-10-24:
Hello，
Please provide the screen resolution you use. Will the font loads correctly if you lower the screen resolution?

khmewes, 2023-10-24:
Hi,
I am using  4096x2160 screen resolution. But I think the issue is caused by text scaling if I go greater than 300% in the Windows settings. At least 300% is needed to be able to read any text from 6-8 feet away ( I am using a 50 " TV as a monitor). But to accomodate my prebyopia I used to set it at 350% which is more readable. So at 350% text scaling that's when the font became unreadable. as shown. It would be great if I could use 350% scaling. Thanks.

LiZ-Li-BBL, 2023-10-30:
Sorry that we did not adapts the high resolution screen/high scale rate well. We recorded this issue and we will fix it as soon as possible.

khmewes, 2023-10-30:
That’s great, thank you. Looking forward for the fix.

 

Klaus

khmewes, 2023-10-31:
Thank you. Please let me know when you have an update and where to download it.

 

Klaus Mewes

khmewes, 2023-11-14:
Hi, did you have a chance to look at this issue yet? When can I expect see the fix in a new release of the software?

 

Thank you.

 

Klaus Mewes

khmewes, 2023-12-20:
Hi,

You have released v1.8.2.56 but unfortunately you didn’t fix this resolution problem yet. Is it still under consideration?

 

Thank you 

Klaus Mewes

lllucius, 2023-12-28:
In case it helps, I'd also have this problem.  I always run at 350% scaling due to eye sight issues and hope to see a fix soon.  Thanks.

khmewes, 2023-12-28:
Thanks Illucius for backing me up here. Bambu Lab has recently released a new version of fixes, but have not addressed this problem yet. I guess it is low on the totem pole as far as bugs go. I really hope they will fix this soon. I don't know how much of a problem this can be? Adding larger font support or scaling? Not sure.

lllucius, 2023-12-28:
Well, I found a workaround that works for me, maybe it'll work for you as well.

1)  Locate or create a shortcut for BambuStudio
2)  Right click it and select "Properties"
3)  Select the "Compatability" tab
4)  Click the "Change high DPI settings" button near the bottom
5)  Check the "Override high DPI scaling behavior..." checkbox near the bottom
6)  Select "System (Enhanced)" from the dropdown (default is Application)

khmewes, 2023-12-28:
Thank you very much for figuring this out. I works for me now too. 350% is enough to make things readable. No more blocky, unreable font in the preview panel. Great job.

khmewes, 2023-12-29:
Bad news though. The 'sliders' on the left panels and other sliders didn't respond anymore. Which makes it useless. So I went back to default comp settings and 300% scaling. Did you notice this also? Or was I doing something wrong in the compatibility settings?

lllucius, 2023-12-29:
Dangit!  Same thing here too.  I'm just getting into this whole 3D printing thing (still waiting for the printer) and didn't even check the other controls.  Sorry about that.

So, I guess we're back to waiting.

lllucius, 2023-12-29:
Maybe good news.  Try the "System" option instead of the "System (Enhanced)" one.  I've just tried it and, if I'm seeing the same thing you're seeing, then this "may" be what we need.

khmewes, 2023-12-30:
Looks like that works. I used the the 350% setting, the only option after 300% (recommended) and the slider work now and the font is nicely readable. The only thing is, with the font being larger 'everywhere' the work space with plates is getting smaller. Also the icon for 'Assembly View' is cut in half on the 'tools' bar. Still works though. So there might be some other issues with cropping of panels in the device/printing screen, like seeing all the option for filament parameter editing panel. I had problems there even with 300% before. Panels cut off and couldn't be moved. But it's a nice fix/work-around. Thanks again and Happy New Year.

khmewes, 2023-12-30:
Quick followup. There is a downside. Everything seems to be 'rough around the edges.
Fonts, icons, models, plates etc. So we still need Bambu Labs real fix, if possible.

lllucius, 2023-12-31:
An interesting development in the OrcaSlicer:

[QoL: Make sidebar collapsible, resizable, movable, floatable and dockable](https://github.com/SoftFever/OrcaSlicer/pull/2972)

I know that OrcaSlicer is based on Bambu Studio, but I have no idea if it's better or worse.  But, I did give the latest beta a try and this feature helps me A LOT!  At 350% scaling, I might a bit off the right side of the toolbar and most of my display is used by the sidebar (I only have 3840x2160 res).  Being able to reside, float or hide the sidebar is going to be a great feature.

Unfortunately, it still has the same text problem at 350%.  I have built it locally and "hope" to find what needs to be changed to fix that, but it's a big project and I may not be able to figure it out.

lllucius, 2023-12-31:
I have been successful at getting the text to display correctly in Orca Slicer.  If I can ever get Bambu Studio to build (it's being difficult), I should be able to do the same thing there.

It's a small change, but I don't know how "major" a change it is as I don't currently have Mac or Linux machines set up to test it there as well.  Best I could do is submit a pull request and see if it's accepted.

lllucius, 2023-12-31:
Works well in Bambu Studio as well.  I'll submit a pull request and we'll see if the change is acceptable.

khmewes, 2024-01-01:
Good job. I am not as experienced in programming as you are, Far from it. But I appreciate your effort. I will wait for the official 1.9 release of Orca with your changes for downloading. Also hoping that BL will accept your changes and release them in a future version.
Thanks and Happy New Year 2024.

lllucius, 2024-01-01:
If you want to see what it looks like with the fix, check out the Orca issue:

https://github.com/SoftFever/OrcaSlicer/issues/3401

If you want to try it, they have an automated build set up when someone submits a pull request.  You can try the result of that build here:

https://github.com/SoftFever/OrcaSlicer/actions/runs/7374535313/artifacts/1141636147

khmewes, 2024-01-04:
Hi,

Thanks for the link.

I tried version 1.9.0 beta and like the show/hide left panel. However I am using dark mode and the print plates don’t show in dark mode, everything is grey except for the model objects. Did you notice that as well? Regular mode seems fine.

 

Best Regards,

 

Klaus

 

SoftFever/OrcaSlicer#3401 <https://github.com/SoftFever/OrcaSlicer/issues/3401> 

https://github.com/SoftFever/OrcaSlicer/actions/runs/7374535313/artifacts/1141636147

lllucius, 2024-01-05:
> Hi, Thanks for the link. I tried version 1.9.0 beta and like the show/hide left panel. However I am using dark mode and the print plates don’t show in dark mode, everything is grey except for the model objects. Did you notice that as well? Regular mode seems fine. Best Regards, Klaus 

I use dark mode as well and Bambu Studio and Orca Slicer (1.9.0) both display properly with the fix and 350% scale. Did you grab the 1.9.0 version from the artifacts in the pull request or from the officlal builds?

khmewes, 2024-01-05:
I think it was the official 1.9.0 beta.

Can you send me the link for the version you are using please?

I also thought I read somewhere that a new version will allow to ‘tear-off’ the panel and place it anywhere. And there is a version 1.9.1 now.

 

K.

lllucius, 2024-01-05:
> Can you send me the link for the version you are using please?

Here's the build from the pull request:

https://github.com/SoftFever/OrcaSlicer/actions/runs/7374535313/artifacts/1141636147

It's still the 1.9.0 fella, but it'll let you see if you still have that "grey" problem.

khmewes, 2024-01-05:
Ok thanks. This version is working. I tried 350% and no problems.

 

https://github.com/SoftFever/OrcaSlicer/actions/runs/7374535313/artifacts/1141636147

khmewes, 2024-01-07:
Hi,

I have tried the release version 1.9.0 and I discovered a display bug in the notification area that the beta (your link below) does not have. The text is missing in the release version (bug?).

Screenshots are included.

 

https://github.com/SoftFever/OrcaSlicer/actions/runs/7374535313/artifacts/1141636147

lllucius, 2024-01-07:
> I have tried the release version 1.9.0 and I discovered a display bug in the notification area that the beta (your link below) does not have. The text is missing in the release version (bug?).

They haven't accepted my changes as yet (and may not).  Even though the change is quite small, the impact could be large.  One of the critiques was that by changing the way the fonts are scaled, blurriness can occur on some platforms.  That prompted me to build a Linux box and I did indeed see a significant blurriness when my change was in place.

I updated the change and made it Windows specific (as suggested) in both Bambu Studio and OrcaSlicer and that's where we currently sit.  I've gone through as much of the UI as I can without actually having a printer (being delivered tomorrow), so I can't say with 100% certainty that it doesn't break anything.

So, we're still waiting for it to be included in the releases.  If they decide to not include it, I'll just have to maintain a forked version with just this fix applied and try to keep up with the upstream releases.  Not ideal, but my bad eyeballs have caused me worse grief than this over the years.  😃

Though, I can't believe it's only me and you that have experienced this issue...

PR #3271 (2024-01-08): Fix garbled text when display scalling > 300%
On Windows, if your display scaling is greater than 300%, the text rendered by imgui will be garbled. Allowing imgui to do the scaling instead of making the font to large corrects the issue.

This fixes #2626
closed 2024-01-08 by LiZ-Li-BBL