bambulab/BambuStudio #2627: Support infill is generated 2x normal density

tsmith35, 2023-09-28:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: 13900K
GPU: RTX 4080
RAM: 64 GB

### Printer

Bambu Lab X1C, P1P

### How to reproduce

Create an object with infill. Select "support cubic" for infill type. Slice object and view preview.

### Actual results

Infill generates at 2x the expected density. For example, 20% support cubic infill generates at 40% fill.

### Expected results

Infill should be generated at the expected 20% density. To work around issue, set fill to 1/2 the expected density. i.e., select 10% infill with support cubic to generate the expected 20% density. Only support cubic appears to be affected.

This issue affects both OrcaSlicer and Bambu Studio. Tested against PrusaSlicer 2.6.1 and Cura 5.4.0

Project file & Debug log uploads

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-09-28:
could you provide 3mf file?

tsmith35, 2023-09-28:
Sure!
[Infill Examples (adaptive & support).zip](https://github.com/bambulab/BambuStudio/files/12745426/Infill.Examples.adaptive.support.zip)

QingZhangBambu, 2023-10-16:
[image]
i try it, adaptive cubic and support cubic excatly same. what's your steps? i tend to reproduce it.

tsmith35, 2023-10-16:
> !i try it, adaptive cubic and support cubic excatly same. what's your steps? i tend to reproduce it.

You just did. The support cubic should appear to be twice the scale, but it's not. Support cubic is just like adaptive cubic but is exactly half as dense.

QingZhangBambu, 2023-10-24:
ok , i will check it

QingZhangBambu, 2023-10-26:
the density of the support cubic infill relates to the distance to the top. if the height increases, the support cubic density is smaller than the adaptive cubic.
[image]

QingZhangBambu, 2024-01-17:
Closed due to no reply

closed 2024-01-17 by QingZhangBambu