bambulab/BambuStudio #2629: Bambu Labs Studio unable to send plate

Jpeterson37, 2023-09-28:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

Slice a file
select send instead of print plate
click send
click send

### Actual results

Error message pops up to confirm IP and LAN access key
Confirm and states it is incorrect

### Expected results

Sends file over LAN so it can be printed at the printer. Attempting to stage large file prints. The printer is not in LAN only mode. When in LAN only mode I get a time out error for FTP server.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12744622/log.zip)
Project file to large to upload.

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-10-01:
HI. @Jpeterson37 
When a dialog box pops up to confirm the IP and access code, it indicates that there are some issues with FTP transmission. You can first check that your printer network or Studio network is working properly. Try sending a small file to confirm if the transfer failed due to the file being too large.
If the file is too large, it is recommended to use an SD card for copying.

Jpeterson37, 2023-10-02:
Hello. I believe I found the issue to be related to BSS Transition on the wireless  network used. I haven't seen any issues since disabling. Does the WIFI support any advanced WIFI features.

walterwongbbl, 2023-10-07:
Hi. @Jpeterson37 
We only use the most basic network functions and don't configure any wifi options. Is the issue resolved now?

Jpeterson37, 2023-10-07:
This is resolved