bambulab/BambuStudio #2630: File model keeps shifting layers

David3333333333333333, 2023-09-28:
### Bambu Studio Version

1.7.6.9.2

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz   2.30 GHz
RAM 32gbs
NVIDIA 3060

### Printer

BAMBU LAB X1C

### How to reproduce

basically i sliced a specific model and it keeps layer shifting. The model file name is helmet split top left.

### Actual results

in the middle of the print a specific layer always shifts but the slicer does not show that in the preview.

### Expected results

you get an ugly print job.

### Project file & Debug log uploads

[ShredderHelmet2STLs-EXTRACTTOUSE.zip](https://github.com/bambulab/BambuStudio/files/12746215/ShredderHelmet2STLs-EXTRACTTOUSE.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-10-03:
@David3333333333333333 

is this happens when printing other 3mf?

meanwhile,
as the develop team is on holiday
we are sorry for the slow response before 7th, October

SaltWei, 2023-10-07:
[image]
Can not find any abnormal shift in gcode. I guess the motor of printer loss step by coincidence during printing.
Could you share one picture of printed model to show the position of layer shift?

David3333333333333333, 2023-10-07:
[image]

David3333333333333333, 2023-10-07:
> @David3333333333333333
> 
> is this happens when printing other 3mf?
> 
> meanwhile, as the develop team is on holiday we are sorry for the slow response before 7th, October

no it doesn't just this model

SaltWei, 2024-01-18:
@QingZhangBambu Please reprint this gcode and check whether it can be reproduced.

QingZhangBambu, 2024-10-15:
i print the model and there is no shift layer.