bambulab/BambuStudio #2633: X1C - Device - Crashes on Model in SD Card window

jason-e-gross, 2023-09-28:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

CPU: 12th gen I7, 64GB DDR3600, 2TB SSD

FAT32 formatted 32GB SD card in X1C

### Printer

Bambu X1C

### How to reproduce

1. Go to Devices
2. Select your X1C printer
3. Go to SD Card tab
4. Look at Timelapse stuff
5. Click "Year" 
6. Click "Model" tab on the right

Crashes

### Actual results

It crashes and the app closes

### Expected results

The app doesn't crash or close

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12750274/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jason-e-gross, 2023-09-28:
Note - I have to click the "Year" tab in the timelapse top bar first.      If I don't click anything going into SD Card, and move right to Model - it just shows "Nothing Here".   It's when I interact with the Year | Month | All Files section before going to "Model" is when it bombs.

bambu123, 2023-10-07:
It is indeed a permanent problem, we will modify it ASAP.

bambu123, 2023-10-24:
Fixed!

closed 2023-10-24 by bambu123