bambulab/BambuStudio #2635: Bambu Studio shows incorrect status of printer if computer hibernates during print

ThorsReign, 2023-09-28:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Core i7-7700HQ
32 GB RAM
NVidia GTX1060

### Printer

Bambu P1S w/AMS

### How to reproduce

1. Start print of project
2. Put computer in hibernation mode during printing
3. Allow print to finish
4. Wake computer from hibernation

### Actual results

Device tab in Bambu Studio shows the status bar of the print at the point where the system hibernated, and the camera feed shows the live feed where the print is finished.  Attempting to print a new project states the printer is busy even though the printer itself and Bambu Handy show the print as completed and successful.  Trying to refresh from the print dialog does not update anything, trying to stop the print from the device tab gives the confirmation dialog but doesn't accomplish anything.  This happens for every project used regardless of print settings (have replicated with 4 different prints and 4 different projects.)

### Expected results

The connection to the printer via LAN or Cloud should accurately read the current printing state of the printer and update to show that the print was completed and allow you to start a new print.  You have to exit and reload Bambu Studio to clear the Device page and begin a new print.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12754414/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2023-10-03:
should be similar issue as https://github.com/bambulab/BambuStudio/issues/2651

walterwongbbl, 2023-10-07:
Hi. @ThorsReign 
Sorry for the inconvenience. We are investigating the issue, where if a connection reset occurs when printing is nearly completed, it results in the loss of print progress. We are working on fixing this problem. For now, you can resolve the issue by restarting Studio.

closed 2023-11-24 by walterwongbbl