bambulab/BambuStudio #2636: Incorrect wording after load filament selected

itsonlymedan, 2023-09-28:
### Bambu Studio Version

1.7

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

P1 Series

### How to reproduce

After the load filament button is selected and the printer begins the process. 

### Actual results

The Printer asks if the user is Done or if they need to retry. This is correct. 

The Slicer asks if it is Done or Confirm 

### Expected results

The Slicer should ask Done or Retry

### Project file & Debug log uploads

```        if (it_done != message_containing_done.end()) {
            m_print_error_dlg->update_func_btn(_L("Done"));
            m_print_error_dlg->update_title_style(_L("Warning"), SecondaryCheckDialog::ButtonStyle::CONFIRM_AND_FUNC, this);
        }
        else if (it_retry != message_containing_retry.end()) {
            m_print_error_dlg->update_title_style(_L("Warning"), SecondaryCheckDialog::ButtonStyle::CONFIRM_AND_RETRY, this);
        }else {
            m_print_error_dlg->update_title_style(_L("Warning"), SecondaryCheckDialog::ButtonStyle::ONLY_CONFIRM, this);
        }```

### Checklist of files to include

- [X] Log file
- [X] Project file

xperiments, 2023-09-28:
Actually the problem is located in the json you are providing at https://e.bambulab.com/query.php?lang=en.
We have only checked EN but may be affecting others also.
The affected data is in device_error "ecode":"07FF8007"

fdebeer, 2023-09-29:
2nd this. also translation in dutch language is wrong

cwr5033, 2023-09-29:
New user, this was VERY confusing at first to understand "confirm" means "retry".

bam-winterfell, 2023-10-16:
The new version has resolved this issue, please update it promptly. Thank you for your feedback. Wishing you a pleasant life

closed 2023-11-14 by bam-winterfell