bambulab/BambuStudio #2639: New studio update broke load button

CadDaddy42, 2023-09-29:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

HP laptop with AMD A8 

### Printer

P1S

### How to reproduce

Open device tab
Click 'load' button below filament display with spoolholder source selected (AMS greyed out because it's unplugged to use filament on a cardboard spool)

### Actual results

Nothing happens

### Expected results

Printer initiates load filament routine

Unload button in studio still works. Load button worked earlier today. Updated firmware yesterday, and studio today - now the load button in Studio doesn't work anymore (first filament change since updating studio, 3 filament changes since updating firmware). Tried turning it off and back on again (everything - printer, slicer, and laptop). Load function works fine when selected from the UI on the machine. 

### Project file & Debug log uploads

Log file attached
[log.zip](https://github.com/bambulab/BambuStudio/files/12756476/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

CadDaddy42, 2023-09-30:
New issue noticed - it hung up during unload, after pulling the filament out it continued to read the status as if the filament sensor didn't notice I removed the filament. I had to shut down both studio and the printer and restart them to clear that, then it recognized there was no filament and I was able to load new filament via the load command on the machine control screen menu.  I never had this issue before these last updates, either.

nistei, 2023-10-01:
Can confirm, nothing happens.

walterwongbbl, 2023-10-12:
Hi.  @CadDaddy42 @nistei 
https://github.com/bambulab/BambuStudio/issues/2699
You need to first select the tray you want to operate on, and then proceed with the load operation.

closed 2023-12-21 by walterwongbbl