bambulab/BambuStudio #2645: Switching from A1 mini printer to X1 does not correctly place objects when using multiple print plates

kadrim, 2023-09-30:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 series

### How to reproduce

1. Visit https://makerworld.com/en/models/14265#profileId-14275 or use the attached 3mf file
2. Open in Bambu Studio
3. The project will be opened in bambu studio with the A1 mini print profile
4. switch to a bigger printer i.e. X1

### Actual results

Opening a file that has been prepared for the A1 mini (with a slightly smaller print area) and with multiple plates involved will not correctly place the objects when switching to a bigger printer like the X1.

The objects will always be placed according to their global coordinates instead of the relative plate coordinates

Screenshot 1 - Opening the file (A1 mini profile):
[image]

Screenshot 2 - Switchted to a bigger printer (X1 profile):
[image]

### Expected results

switching from a smaller printer to a bigger printer with multiple plates involved should place all objects correctly on the bigger print plates

### Project file & Debug log uploads

[Marble Run Components Kit-003 beta5.zip](https://github.com/bambulab/BambuStudio/files/12774191/Marble.Run.Components.Kit-003.beta5.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-10-03:
thanks for this advice, this is reasonbale

we will improve it

StoneLiBambulab, 2023-10-18:
Hi @kadrim 

it will be added in release 1.8

Thanks!

Stone Li

SaltWei, 2024-01-17:
Already included 1.8.2 version on github and official website

closed 2024-01-17 by SaltWei