bambulab/BambuStudio #2646: Edit Text command resets any Object specific settings

GuyH77, 2023-09-30:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

OS Name	Microsoft Windows 10 Enterprise
Version	10.0.19044 Build 19044
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	PC019365
System Manufacturer	HP
System Model	HP ZBook Fury 15 G7 Mobile Workstation
System Type	x64-based PC
System SKU	9VS27AV
Processor	Intel(R) Core(TM) i9-10885H CPU @ 2.40GHz, 2400 Mhz, 8 Core(s), 16 Logical Processor(s)

### Printer

Bambu Lab X1C 0. nozzle

### How to reproduce

1. Create a text object on a part
2. Apply Object specific settings e.g. speeds, walls, infill,
3. Use Edit Text command, change the text. 
4. Check the Object settings. They will have reset to the default.

[image]
Original Text, Object settings applied.

### Actual results

If you use Edit Text. make and edit, the Object Settings are reset to default as if it is new text. It is still the same object and therefore I would expect the settings to remain.

Following an edit.
[image]

### Expected results

Following an edit I would expect all object settings to remain as they were.

### Project file & Debug log uploads

[TEXT_TEST.zip](https://github.com/bambulab/BambuStudio/files/12775411/TEXT_TEST.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

GuyH77, 2023-11-10:
@StoneLiBambulab Just testing the 1.8 beta and this bug is not fixed.

BambooBugFixer, 2023-11-16:
Hi, thanks for reporting this. This bug will be fixed soon.

elsni, 2024-02-01:
It also resets the position of the text if you have moved it. Very annoying.

GuyH77, 2024-02-05:
> Changing the text also resets the position of the text if you have moved it along the Z axis after creation. Very annoying.

That's another good point, any change to the text resets the position, even in X and Y. When I compare to Prusaslicer they handle it a lot better in this regard. Definitely can be improved.

GuyH77, 2024-04-29:
@BambooBugFixer can the text tool be improved such that any change doesn't reset the anchor point/position of text. The anchor point is inaccurate and only serves the purpose of defining the surface for the text. Any precise manipulation you then do with the move command is lost as soon as you edit the text.

seanbo, 2025-09-29:
> We will close this issue in one week. If you think this issue still needs to be resolved, please reply within this period

It still needs to remain open.  I have this issue in v2.2.2.56