bambulab/BambuStudio #2648: Excessive bed leveling

GuzziRaz, 2023-09-30:
When flow calibrating many filaments, you really grow tired of the full bed levelling performed each time for no reason. If I didn’t select a different build plate and it did a full mesh 10 minutes ago, why do it again? It should be optional 

See also #2465

UnglazedSquash, 2023-10-02:
The method they are using to implement bed leveling requires them to do it each time. Just the name of the game. Plus it is the safest option.

GuzziRaz, 2023-10-02:
Are you saying this somehow can’t be improved on?

UnglazedSquash, 2023-10-02:
I wouldn't never say you can't. I'm just saying the current time investment to improve on it is probably not justifiable given the results currently being had with the current method. In the future I'm sure it will be improved on. Just maybe not the highest priority item at the moment for them.

magnumripper, 2023-10-02:
Time investment? BS, it's a trivial fix and it would save me and thousands of other users lots and lots of exactly that: **Time**.

GuzziRaz, 2023-10-02:
Your comments make no sense and add no value to this issue.

SaltWei, 2023-10-07:
> If I didn’t select a different build plate and it did a full mesh 10 minutes ago, why do it again?

Since after the hot bed removed from the magnetic plate and then reinstalled, the horizontal state may also change.

SaltWei, 2023-10-07:
By the way, if you pay attention to the bed level process, you can find that the printer is probed 5 points first to check whether bed level has been changed. If not changed, then it will not do the full bed level.

GuzziRaz, 2023-10-10:
OK that's obviously acceptable, except my bed seems to do the full program every single time. Between those calibrations, I just carefully scrape the previous calibration off without removing the plate. Does this mean my bed is somehow unstable?

GuzziRaz, 2023-10-13:
Paying closer attention to the process, you are right it’s not always doing a full ABL. Calibration of 8+ filaments after a nozzle change is still tedious but the main problem could be solved by #2465 so I’m closing this.

closed 2023-10-13 by GuzziRaz