bambulab/BambuStudio #2650: Filament "load" button in BambuStudio is no working

MarkB2023, 2023-10-01:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

P1P

### How to reproduce

1. i reload the old filament --> its working !
2. i will load the new filament with the button in the slicer --> it dosent work. Only with the "Load" Button in the printer display is working
3. in the last update all was wonderfull....
4. thanks

### Actual results

it dosent work

### Expected results

it dosent work

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12777027/log.zip)
[Gwurzschale.zip](https://github.com/bambulab/BambuStudio/files/12777028/Gwurzschale.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-10-07:
@MarkB2023 
Have you used AMS? You can first select the tray you want to load, and then click the 'Load' button.

Dartanan, 2023-10-09:
I've been experiencing the same issue since the update, I've been using Unload button instead, since it heats up, cuts the filament, keeps the high temperature (250C), prompts to unload the filament, after detecting it was removed the extruder is slowly pulling in any incoming filament for couple of seconds and then changes the temperature to 210C. This is all very different behaviour then previously. Load button however does nothing at all now. P1P without AMS, I don't have AMS.

walterwongbbl, 2023-10-12:
@Dartanan 
https://github.com/bambulab/BambuStudio/issues/2699

walterwongbbl, 2023-10-12:
@Dartanan 
Hi. You need to click on the slot first, and then click on the load button.

closed 2023-12-21 by walterwongbbl