bambulab/BambuStudio #2651: Bambu Studio Doesn't update printer status

kazolar, 2023-10-02:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu X1C - FW:1.06.05.01

### How to reproduce

Start a print in bambu studio, leave, let the PC go to sleep. Come back when the print is done. Bambu studio connects to the printer shows the print still in progress, no updates -- printer still shows printing temps.

Doesn't always accept new bed temp (to preheat for high temp plastic) -- combination of new firmware and new bambu studio has started this behavior. Will try to revert X1C firmware. New firmware is very glitchy. 

### Actual results

Can't start a new print until forcing ghost print stop -- it then updates to current printer readings. 

### Expected results

Expect the bambu studio to show current printer status accurately (as it did before)

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12779903/log.zip)
[Ducts.zip](https://github.com/bambulab/BambuStudio/files/12779908/Ducts.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GuzziRaz, 2023-10-02:
An even worse variant of same problem is when you start a calibration from Bambu Studio and then go do something useful during it. If computer had went to sleep when you get back (and calibration is done), calibration screen shows old data as in not finished. You're lucky if you can get Studio to recognize that calibration is done so you can enter the new values. I really hate it when a perfectly fine calibration has to be done again just because there were no way to save the data.

lanewei120, 2023-10-03:
so the issue is related with PC's sleep?
@kazolar 
could you help to mention out the extract sleep time and wakeup time in your log?

thanks very much

meanwhile,
as the develop team is on holiday
we are sorry for the slow response before 7th, October

walterwongbbl, 2023-10-07:
@kazolar 
Sorry for the inconvenience. We are investigating the issue, where if a connection reset occurs when printing is nearly completed, it results in the loss of print progress. We are working on fixing this problem. For now, you can resolve the issue by restarting Studio.

kazolar, 2023-10-07:
@walterwongbbl restarting studio doesn't fix it. I have to restart studio and power cycle the printer

kazolar, 2023-10-13:
I rolled back the printer firmware using the Handy app, and switched to Orca Slicer. Issues are not present in this combination.

lanewei120, 2023-10-15:
> I rolled back the printer firmware using the Handy app, and switched to Orca Slicer. Issues are not present in this combination.

if only switch back to the old printer firmware using bambu studio, is the issue still there?

closed 2023-11-24 by walterwongbbl