bambulab/BambuStudio #2653: Bambu Studio poor camera quality

KuraiKitsune, 2023-10-02:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 22H2

### Additional system information

CPU: Intel i7-12700K
RAM: Corsair Vengeance 64GB (4x16GB) DDR4
GFX: MSI RTX 4090 SUPRIM 24GB
Connected to network via Gigabit ethernet

### Printer

Bambu Lab X1C

### How to reproduce

Go to Device tab. Press Play on the video window.

### Actual results

The video quality and fps has always been terrible on the Bambu Studio app since the Kickstarter units went out, and has never improved for me.
I do not use LAN-only mode as I appreciate the cloud connectivity.
However, the Bambu Handy app on an android phone on the same home network gets full video quality and fps. The difference is night and day.
Why can they be so different on the same network?
It even looks better on my phone when connected via 4G.

Below are 2 videos of the same print job happening, viewed from the 2 apps.

Bambu Studio: https://i.gyazo.com/db5b19a1b52215133ad94298a324a3a3.mp4
Bambu Handy: https://i.gyazo.com/a2873492e3ff3c684f131d83f93b070e.mp4

### Expected results

Clear, high frame rate video like the mobile app gets.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12785289/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

KuraiKitsune, 2023-10-02:
I connect my phone to 4G only, and the video feed in Bambu Handy is butter smooth.
I connect my phone to Wifi in the home, printer is connect to the same wifi, video feed is butter smooth.

I installed Bambu Studio on 2 different Windows computers, both Win11 and Win10. Both on wired connection to the same network. The video feed from Studio is slow and grainy.
This is how it has always been since I got my X1C from the Kickstarter campaign in the beginning.

bambu123, 2023-10-07:
I think the difference is caused by the size of the video window. The same video will definitely appear poor quality on a large window. On the other hand, it is possible that the mobile phone system has a video image enhancement module, while the Windows platform does not have such a module by default.

closed 2025-10-31 by KuraiKitsune