bambulab/BambuStudio #2656: Trying to slice the Hands Of Ganon causes BambuStudio to crash

djengis, 2023-10-03:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Microsoft Windows 11 Pro

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i9-13900K, 3000 MHz, 24 Kern(e), 32 logische(r) Prozessor(en)
MB: MPG Z790 CARBON WIFI (MS-7D89)
RAM: 32GB DDR5 DDR5-6400 DIMM Corsair
GPU: 12GB Gigabyte Geforce RTX 4070 Gaming OC
PSU: 850W
Monitor: QHD with 144Hz FreeSync

Software:
GPU-Driver: GeForce Game Ready Driver 537.42

Settings:
- PEI Plate
- Bambu PLA Basic
- 0.12 Fine
- Sparse infill denstiy: 10%
- Enable Support -> Tree (auto)

### Printer

Bambu Lab P1P

### How to reproduce

Load attached file in the current version of Bambu lab. 
Ensure the settings are as described via Additional system information.
Try to slice it -> CRASH.

### Actual results

Most of the time the slicer crashed without any notice. Its just gone.
Sometimes my whole system crashed and the file gets unreadable.

### Expected results

Beautiful G-Code like G-od intended. :)

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/12792890/logs.zip)
[crashing--ganon-hands-zelda-totk (2).zip](https://github.com/bambulab/BambuStudio/files/12792908/crashing--ganon-hands-zelda-totk.2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-10-07:
Can not reproduce in my win10 PC. Keep investigating

SaltWei, 2023-10-07:
Seem to be dead in 3-rd party lib code.
[image]

XunZhangBambu, 2023-10-13:
Also cannot reproduce in my win11 PC

djengis, 2024-01-01:
Finally found the issue. It seem it was a hardware problem. MSI's auto overclock of the CPU made these errors appear. Turning it of fixes this. Strange that i never had an issue in any game or other software like blender.

closed 2024-01-01 by djengis