bambulab/BambuStudio #2658: support surface filament used on non-surface layers of the  support structure

min-xu-ai, 2023-10-03:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux

### Printer

X1C

### How to reproduce

See the question here: https://forum.bambulab.com/t/using-pla-as-support-interface-and-odd-slicing/30130/5

I have exact same issue as seen in this slicer output:

[image]

### Actual results

The green filament is used on the support body for some layers where the support interface is located. This causes the support to be unstable since the interface layer is supposed to be only on the interface.

### Expected results

I'd expect only the interface portion of the support is printed with the interface material. That means the filament needs to be switched in the middle of the layer when printing the interface vs. the body of the support.

Am I mis-understanding something?

### Project file & Debug log uploads

[Cable_and_earbud_clipers.3mf.gz](https://github.com/bambulab/BambuStudio/files/12795216/Cable_and_earbud_clipers.3mf.gz)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-10-07:
[image]
The support body filament is default, which means the current used filament. This can reduce the filament change times.
You can set it to specific filament to avoid using interface filament.

min-xu-ai, 2023-10-07:
Interesting! I didn't thought about that. Will test it out. Thanks for your help!

min-xu-ai, 2023-10-11:
I tested this and it works once I changed support/raft base from default to the right filament. I think this can be closed. Please reopen if I shouldn't have closed it.

closed 2023-10-11 by min-xu-ai