bambulab/BambuStudio #2659: Evacuate Chamber option

the-robo, 2023-10-03:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to manually start the aux and chamber fans to evacuate air after a print job is complete. To get to the condition I favor I have to CLICK 20 TIMES.

**Describe the solution you'd like**
I would like a button that is similar to chamber fan button. Once clicked a screen pops up that allows me to configure how I would like the air in the chamber evacuated. Once configured the settings would be retained so during normal use I would only have to click once or two times at most.

**Describe alternatives you've considered**
There are no alternatives currently. The only option is to click 20 times in the current interface.

**Additional context**
I have printed an adapter which allows me to connect a vent hose from the chamber to an outside vent. This allows me to evacuate the chamber of any potential particles remaining after a print job. This also helps cool the engineering or other hot plate to manageable levels.

GuzziRaz, 2023-10-04:
> configure how I would like the air in the chamber evacuated

The configuration could include a timer setting as well ("run fans for *n* minutes").

In addition to the touchscreen button for running this, a print settings checkbox for it in the slicer would be nice as well. When checked it would automagically run the same "Evacuate Chamber" configuration right after print  is finished.

GuzziRaz, 2023-10-04:
> The only option is to click 20 times in the current interface.

Until this is implemented you could append some custom commands to "Machine End G-code". This should do the trick:
```
M106 P2 S180 ; Set aux fan to 70%
M106 P3 S255 ; Set chamber fan to 100%
M400 S600 ; Pause for 10 minutes
M106 P2 S0 ; Turn off aux fan
M106 P3 S0 ; Turn off chamber fan
```
A variation of it could replace the static pause with "wait until nozzle temp [and consequently chamber temp] has decreased to 30°C". It could be accomplished with `M109 R30` if Bambu honors that Marlin gcode. It's on my to-do list to try that.

the-robo, 2023-10-04:
> M106 P2 S180 ; Set aux fan to 70%
> M106 P3 S255 ; Set chamber fan to 100%
> M400 S600 ; Pause for 10 minutes
> M106 P2 S0 ; Turn off aux fan
> M106 P3 S0 ; Turn off chamber fan

This sounds like a good work-around, I will give it a try. The biggest drawback I can think of is having to configure each filament I want to use this for.  A minor drawback is I may not want it to run every time, hence the ad-hoc button option.  Still, I like the way you think! Thanks for this.

the-robo, 2023-10-04:
> The configuration could include a timer setting as well ("run fans for _n_ minutes").
>
> In addition to the touchscreen button for running this, a print settings checkbox for it in the slicer would be [...]

Yes please!

gustafssone, 2023-10-10:
I can confirm I have the same issue. It works with Machine G-code but not with filament g-code.
; Filament gcode
M106 P3 S50 ; Run chamber fan at 20%

Machine start G-Code:
;==CUSTOM for ABS/ASA
{if (filament_type[initial_no_support_extruder]=="ABS")||(filament_type[initial_no_support_extruder]=="ASA")}
         M106 P3 S25 ; Run chamber fan at 10%
{else}
         M106 P3 S0 ; turn off chamber fan
{endif}  ; exhaust VOCs from chamber for ABS/ASA filament

This will set the chamber fan to only 10%, meaning filament g-code doesnt work (I checked the g-code, and it is there, just not working).

I have a P1S.

Oh, and you can have e.g. 
M190 R50 ;wait for bed temperature 50 degrees 
instead of the G4 S300/M400 S300 (wait 5 minutes). That way it cools down to whatever you need.