bambulab/BambuStudio #2662: Studio and Printer fail to warn when sending/printing with incorrect printer.

Jpeterson37, 2023-10-04:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

1. Open link https://makerworld.com/en/models/14267#profileId-23371
2. Open in Bambu Labs Studio
3. By default its set to A1 Mini
4. Slice and send to X1 Carbon
5. No warning on Studio
6. On printer select file and print
7. No warning on printer
8. When printer purges it purges onto build plate
9. When print cuts filament the extruder assembly collides with front panel nearly knocking off fan assembly  

### Actual results

Printer behave unusually and collides with front panel.

### Expected results

Should issue warning or prevent ability to print file that do not match printer type. This will prevent basic users from potentially damaging their printer when using makerworld.com open in Studio feature or other 3mf files.

### Project file & Debug log uploads

https://makerworld.com/en/models/14267#profileId-23371

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-10-07:
@YifanWuBambu Please help to confirm. I think this should not be a bug. Because X1 carbon always can print 3mf of A1 mini profile, right?

Jpeterson37, 2023-10-07:
I found the nozzle assembly would collide with the front left as you look at the printer and knock off the fan assembly

YifanWuBambu, 2023-10-08:
@StoneLiBambulab 
This case need to be blocked when sending the print job.
Please help handle it.

StoneLiBambulab, 2023-10-12:
Hi @Jpeterson37 ,
the printer firmware does not check the printer preset type and physical printer type when printing a new job, causing this problem. I have contacted the printer team to add this checking.

@YifanWuBambu 
这个需要打印机来处理，Studio上发起打印已经增加了类型检查的逻辑，如下图，这个用户遇到的问题是先发送文件到打印机，再从打印机上发起的打印。
[image]

Thanks!

Stone Li（旺仔）

togeis, 2023-12-09:
Same happend to me yesterday. Opened a model from MakerWorld, sliced an send to X1C. X1C purged to plate and crashed into frame!

Afer changing the printer in Bambustudio to X1C, sending to printer, it happens again.

Only after powercycle X1C, changing printer type to X1C, sliced model and newly send to printer helped.

Seams that the A1 settings are kept in the printer memory and only powering off X1C helped.

Regards
Togeis