bambulab/BambuStudio #2663: Unable to run Bambu_Studio_linux_ubuntu_v01.07.06.92-20230923002726.AppImage

mprowe, 2023-10-04:
### Bambu Studio Version

v01.07.06

### Where is the application from?

Bambu Lab github releases

### OS version

openSUSE (close relative to Ubunto)

### Additional system information

Operating System: openSUSE Leap 15.5
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 5.14.21-150500.55.22-default (64-bit)
Graphics Platform: X11
Processors: 32 × 13th Gen Intel® Core™ i9-13900
Memory: 31.1 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: ASUS

### Printer

Bambu 1XC

### How to reproduce

1. Download Bambu_Studio_linux_ubuntu_v01.07.06.92-20230923002726.AppImage and set execute flag.
2. Rename ~/.config/BambuStudio to give clean start.
3. Complete intro options using all the defaults and arrive at Home page (see image below). All "seems" OK at this point.
4. Attempt to log in with user credentials and the application closes without any messages.
5. Attempt to run again, and I can briefly see that I am logged-in before it yet again crashes.

I have been running  Bambu_Studio_linux_ubuntu-v01.07.04.52.AppImage (Aug.17th) for some time without any issues.

### Actual results

[image]
Screenshot at end of setup but before logging in.

### Expected results

Studio should run.

### Project file & Debug log uploads

Project file: None
Debug Log: [debug_Wed_Oct_04_16_27_44_12971.log.0.zip](https://github.com/bambulab/BambuStudio/files/12805272/debug_Wed_Oct_04_16_27_44_12971.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

mprowe, 2023-10-15:
The above problem seems to be continuing to later versions as well.
Tried ver 1.7.07.89. Same problem. Seems to be related to logging-In/security?
From a clean install (no ~/.config/BambuStudio, I can run through the setup (accepting all the defaults)
but as soon as I try to log in, the application crashes.
What did the AppImage have in 1.7.04.52 that is missing in the later Appimages?
Regards, Martin

lanewei120, 2023-10-15:
> v01.07.04.52

we have update openssl from 1.1.1 to 3.1 in recent versions

what is the openssl version on your system?
@mprowe

mprowe, 2023-10-15:
mprowe@localhost:~> openssl version
OpenSSL 1.1.1l  24 Aug 2021 SUSE release 150500.17.15.1
mprowe@localhost:~>

Do you have a guide on upgrading? My OS repo has 3.0.8 available.
But just installing doesn't switch. I guess I need to find out how to (cleanly) uninstall 1.1.1?

lanewei120, 2023-10-15:
how about this one?

https://unix.stackexchange.com/questions/696381/upgrading-openssl-to-version-3-0-2-from-source

mprowe, 2023-10-16:
Thank you.
From that link you sent me, I understand I only need to run openssl-3 instead of openssl-1_1? Not been able to find any advice around the internet on how to (officially) make that change. So this is what I have done:

1. Rename /usr/bin/openssl to openssl-1_1
2. ln -s /usr/bin/openssl-3 /usr/bin/openssl

Now this is my openssl output:
```
localhost:~ # openssl version -a
OpenSSL 3.0.8 7 Feb 2023 (Library: OpenSSL 3.0.8 7 Feb 2023)

platform: linux-x86_64
options:  bn(64,64)
compiler: gcc -fPIC -pthread -m64 -Wa,--noexecstack -Wall -O3 -fmessage-length=0 -grecord-gcc-switches -O2 -Wall -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g -Wa,--noexecstack -fno-common -Wall -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DZLIB -DNDEBUG -D_FORTIFY_SOURCE=2 -DTERMIO -DPURIFY -D_GNU_SOURCE -DOPENSSL_NO_BUF_FREELISTS -DSYSTEM_CIPHERS_FILE="/etc/crypto-policies/back-ends/openssl.config"
OPENSSLDIR: "/etc/ssl"
ENGINESDIR: "/usr/lib64/engines-3"
MODULESDIR: "/usr/lib64/ossl-modules"
Seeding source: getrandom-syscall
CPUINFO: OPENSSL_ia32cap=0xdefa2203478bffff:0x842569
localhost:~ # 

```
Unfortunately, this does not seem to make a difference to running the AppImage. From a clean-slate (no ~/.config/bambustudio), it runs through the setup questions fine. But when I try to log in (using valid credentials) it crashes!
These are the last few lines  when run from the shell.

```
(WebKitWebProcess:10024): Gtk-WARNING **: 08:59:25.238: Theme parsing error: gtk.css:1552:16: '-gtk-icon-size' is not a valid property name
(WebKitWebProcess:10024): Gtk-WARNING **: 08:59:25.238: Theme parsing error: gtk.css:1555:16: '-gtk-icon-size' is not a valid property name
(WebKitWebProcess:10024): Gtk-WARNING **: 08:59:25.251: Theme parsing error: gtk.css:1552:16: '-gtk-icon-size' is not a valid property name
(WebKitWebProcess:10024): Gtk-WARNING **: 08:59:25.251: Theme parsing error: gtk.css:1555:16: '-gtk-icon-size' is not a valid property name
libEGL warning: DRI2: failed to authenticate
Segmentation fault (core dumped)
localhost:/VBShare # 

```

lanewei120, 2023-10-16:
I just built a[ docker image](https://github.com/bambulab/BambuStudio/suites/17263526733/artifacts/986370117), could you also try this image?

mprowe, 2023-10-16:
Sorry.... But no that is not working.
I have tried the #1828 fix (adding the --appimage-portable-home switch [not sure what that does?]) but that didn't help.
I would like (if I knew how) to quote [GuillaumeLeclerc](https://github.com/GuillaumeLeclerc)

> What about an appImage that actually contains everything even if it's large ? I'm sure most people don't care about size and just want something that works.

Not to be too negative, I've attached the shell messages - not that I think that will help! 
[shell.zip](https://github.com/bambulab/BambuStudio/files/12918454/shell.zip)
Regards, M.

lanewei120, 2023-10-16:
> Sorry.... But no that is not working. I have tried the #1828 fix (adding the --appimage-portable-home switch [not sure what that does?]) but that didn't help. I would like (if I knew how) to quote [...]

how about the contents of your  /home/mprowe/.config/BambuStudio/plugins?
you can delete this directory and try again

meanwhile, I will send you a debug library for a try later

mprowe, 2023-10-17:
> you can delete this directory and try again

Yes, I do that every time I try something new.

lanewei120, 2023-10-17:
@mprowe 

please have a try using the attached two [libraries](https://www.dropbox.com/scl/fi/hgrchufdrht23oopgjzcn/debug_networking_libraries.zip?rlkey=z0xlfzddu3fvzp8nilg52zoam&dl=0)

after your crash in previous running, just copy the library into  /home/mprowe/.config/BambuStudio/plugins
and try again

please don't delete previous plugins

mprowe, 2023-10-17:
> and if it still crash with the debug version
could you use the gdb to have a try and catch the callstack for us?

Where is the debug version?

```
mprowe@localhost:~> gdb /AppImages/BambuStudio_ubu64.AppImage
GNU gdb (GDB; SUSE Linux Enterprise 15) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-suse-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://bugs.opensuse.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /AppImages/BambuStudio_ubu64.AppImage...
(No debugging symbols found in /AppImages/BambuStudio_ubu64.AppImage)
(gdb) quit
mprowe@localhost:~> 
```

lanewei120, 2023-10-17:
[image]
the debug libraries have been attached in above comment
@mprowe

mprowe, 2023-10-17:
No. Not the libraries. You (I think) have asked me to run a debug version of Bambu Studio?
I have tried:

1.  **gdb** Bambu_Studio_linux_ubuntu-v01.07.04.52.AppImage
2. **gdb** Bambu_Studio_linux_ubuntu_v01.07.06.92-20230923002726.AppImage
4. **gdb** Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
5.  ... and ...
6. **gdb** BambuStudio_ubu64.AppImage
All report "No debugging symbols found in <*>.AppImage

My apologies for not being able to keep up with your chain-of-thought. I am very keen to help but my linux knowlage is limited. I will need clearer instructions.

lanewei120, 2023-10-17:
I just mean the debug libraried not app image

you can do as follows:
1. copy the debugged libbambu_networking.so into /home/mprowe/.config/BambuStudio/plugins
2. run Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
3. try the steps to reproduce crash
4. if it crashes in 3, then use gdb to try again
5. gdb Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
6. (gdb) run
7. try the steps to reproduce crash
8. get the steps under gdb
9.(gdb) bt
10. paste the result here

mprowe, 2023-10-17:
Took a bit of doing...
But I hope this is useful? One point to note. When running under gdb, it crashed during the download of the network driver.
[image]
Without gdb in play, it crashes during the login process. Hope this helps.

[bambu.zip](https://github.com/bambulab/BambuStudio/files/12931661/bambu.zip)

lanewei120, 2023-10-18:
> Took a bit of doing... But I hope this is useful? One point to note. When running under gdb, it crashed during the download of the network driver. [image] Without gdb in play, it crashes during the [...]

@mprowe 
thanks very much
it seems gdb cause this issue more complex,let's simplify the steps and try without gdb firstly

1. run Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
2. try the steps to reproduce crash
3. copy the libbambu_networking.so from debug directory  into /home/mprowe/.config/BambuStudio/plugins
4. execute following commands
sudo sysctl -w kernel.core_pattern=/tmp/core-%e.%p.%h.%t
sudo ulimit -c unlimited
5. re-run Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
7. check whether it downloading bambu network plugin-in or not(should not download this time)
8. try the steps to reproduce crash
9. after it crashes, there should be a core-dump file under /tmp directory, like core-xxxx
copy the log file and the core to us

the core dump file may have a large size, please use dropfile or something else to share, the github can not upload such a file

mprowe, 2023-10-18:
Step 4. Problems, I'm afraid!
I don't have ulimit compiled on my kernel. Will prlimit be okay? 
If so, can you suggest a command line?

lanewei120, 2023-10-19:
> Step 4. Problems, I'm afraid! I don't have ulimit compiled on my kernel. Will prlimit be okay? If so, can you suggest a command line?

then you can combine step 4 and 5 as
prlimit --core=unlimited ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage

lanewei120, 2023-10-20:
@mprowe 

thanks for your patience~

from the log, I found the attached libraries have a different glibc version which can not be loaded
that is why your network plugin-in still needs to be downloaded

I have found another linux machine to generate the libraries for you
[new debug libraries](https://www.dropbox.com/scl/fi/u47qb6t4efyuf7m1mbbb8/debug_networking_libraries_1020.zip?rlkey=cnxpod98xqfb724othf0c0uwe&dl=0)

you can use these libraries and try again
it should not be blocked by step 6,
if step-6 still needs you to download the image
you can stop the following tests and just send the log to us

mprowe, 2023-10-20:
I am more than pleased to help where I can. But I must say that I am getting very confused and worried that you and I are doing different things. I am building a new VM each time and you may be assuming that historical drivers/configs are in place?

Anyway - the results today are the same as [yesterdays posting](https://github.com/bambulab/BambuStudio/issues/2663#issuecomment-1770266187)
If I've not understood your instruction, give me a new listing based on a clean build.

[core-bambustu_20231020.zip](https://www.dropbox.com/scl/fi/9708cfg9asvcambb7bd44/core-bambustu_20231020.zip?rlkey=nk1s6n538nvj1mmkv5vzwg4ai&dl=0)
[log20231020.zip](https://www.dropbox.com/scl/fi/jmk3sfzdx6v970p143r60/log20231020.zip?rlkey=r1bzszqb561hbg8piduq8i5bg&dl=0)

Regards, M.

mprowe, 2023-10-20:
> the networking library I sent to you worked this time
> however, it seems you didn't try steps 1,2,3 
>
 I thought I did. But ....
> so the application download the networking plugin again and crash at the downloading process
> I also checked the core dump file, it can not be parsed on our side
> so currently you can just try
> 1. copy all the libraries from [linux_01.07.07.07.zip](https://github.com/bambulab/BambuStudio/files/13052420/linux_01.07.07.07.zip) into /home/mprowe/.config/BambuStudio/plugins
> 2. run ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage directly to check the result
> 3. if still has issue, send the logs to us
> 
After logging in, the login popup remains. So I close it. Try to log in again, new popup says "continue to log in".
I say "OK" and it crashes. Try to run again. I briefly see the Home screen with my log in credential before it crashes again.

[log202310201233.zip](https://www.dropbox.com/scl/fi/0ihhkmybvsqofk64dfz9v/log202310201233.zip?rlkey=tqpog48iihvntrkgmnmey01hg&dl=0)

lanewei120, 2023-10-21:
@mprowe 
what is your opensuse version?
we will try to find a similiar linux machine and debug on it

also, what is your gtk_theme?
how about try a different gtk_theme when run bambustudio

for example:
`GTK_THEME=Adwaita ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage`

and another way for a fast try is to run it with sudo privilege
`sudo ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage`

mprowe, 2023-10-21:
>  what is your opensuse version?
>
Listed [Posting #1](https://github.com/bambulab/BambuStudio/issues/2663#issue-1926502256)
> we will try to find a similiar linux machine and debug on it
>
I'm sure that will be much more productive approach rather than trying to train me to press the correct buttons? This is what I'm using: [openSUSE 15.5](https://get.opensuse.org/leap/15.5/#download) Please note that this install is using openssl-1_1.
To comply with your request to use openssl-3, I have run the following steps: [Posting #6](https://github.com/bambulab/BambuStudio/issues/2663#issuecomment-1763951559)
 
> also, what is your gtk_theme?
>
```
mprowe@localhost:~> gsettings get org.gnome.desktop.interface gtk-theme
'Breeze'
mprowe@localhost:~> 

```
> how about try a different gtk_theme when run bambustudio
> 
> for example:
> GTK_THEME=Adwaita ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
>
No different still crashes... 
> and another way for a fast try is to run it with sudo privilege
> sudo ./Bambu_Studio_Linux_ubuntu_v01.07.07.89-20231013104019.AppImage
>
No. Sorry. Still crashing.

lanewei120, 2023-11-09:
Hello @mprowe 

we have just released v1.8.0 public beta version
and we have fixed the login related issue on linux

could you have a try to verify?
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.57

mprowe, 2023-11-09:
Hi,

Thank you for letting me know that there is an update. Very keen to try it out.

But.... Not good news I'm afraid! [Dropbox Link ](https://www.dropbox.com/scl/fi/5a67bvhekpzadlavv4ag3/StudioCrash-2023-11-09_10.41.26.mkv?rlkey=tuzcu00b1ddv4o1ry2oz9f9mx&dl=0)
... [Here](https://www.dropbox.com/scl/fi/a2a1vgfvgo2geeq5eg1h6/BambuStudio.zip?rlkey=ye94zur7pivogmfykjbnznho4&dl=0)

Let me know if there is anything else I can do.

Regards, M.

quantalume, 2023-11-16:
Just a note to confirm that this is also the behavior on my OpenSUSE 15.5 machine. 1.7.04.52 is the latest version working for me, segfault with all the later versions.

mprowe, 2023-12-10:
I notices that [Public beta 3](https://github.com/bambulab/BambuStudio/releases/download/v01.08.01.57/BambuStudio_linux_ubuntu_v01.08.01.57.AppImage) has been released (5/12/2023).
Unfortunately it STILL crashes on openSUSE 15.5 during the Network Plugin download!

Note to Bambu Labs (if they really are listening?):       Please let us help get this working.

lanewei120, 2023-12-14:
I am afraid we have not solved this issue in the coming 1.8 official version also, 
however, we are preparing the OpenSUSE system and try to reproduce it

hope we can solve all the building related issues soon

lanewei120, 2023-12-15:
@mprowe 

we have installed an OpenSUSE system, and can reproduce this issue here
from our debugging, we found it is related with OpenSSL

as we upgraded OpenSSL version to 3.1 before
the function of 'SSL_CTX_new(TLS_client_method());' always return null

it is strange only happens on OpenSUSE
and we can not solve it currently

however, I just build a networking plugin using previous Openssl 1.1.k
which should work at your side
[libbambu_networking.zip](https://github.com/bambulab/BambuStudio/files/13685562/libbambu_networking.zip)

please download this library, 
and extract it into /home/user.name/.config/BambuStudio/plugins directory

meanwhile, we will continue to debug this issue

mprowe, 2023-12-15:
Thank you for keeping me in-the-loop.
I think I have understood your instructions... But I don't think it is working for me.
See attached video:
[Opening Studio ver 01.08.01.57 ](https://www.dropbox.com/scl/fi/sk3k2e3au9fj6hn9db54r/StudioCrash-2023-12-15_14.36.05.mkv?rlkey=hib1vljg5br0sshep57rac09w&dl=0)

klsgh, 2023-12-15:
I unzipped libbambu_networking.zip into ~/.config/BambuStudio/plugins, but every time I start BambuStudio_linux_ubuntu_v01.08.01.57.AppImage it immediately downloads and installs the original version of that file and the crash happens again. Is there anything special I need to do to make it actually use the new version?

lanewei120, 2023-12-16:
@mprowe @klsgh 
sorry, I didn't give the clear steps before

please do as follows:
1. download the public version of 1.8.2 at 
https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage
or 
https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage

2. run this appImage
it should install the network plugins automatically after bootup

3. close the app

4. unzip the above libbambu_networking.zip into  /home/user.name/.config/BambuStudio/plugins

5. bootup the appImage again
this time it should work

mprowe, 2023-12-16:
Sorry @lanewei120,
When I run [1.8.2.56](https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage) it does all the "Get Started" stuff and if I leave the check in "Install Network plug-in" it will crash without creating the **~/.config/BambuStudio/plugin** folder.
If I try again, but don't download the plug-in, it gets to here:
[image]
... and crashes when install it?

lanewei120, 2023-12-16:
> Sorry @lanewei120, When I run [1.8.2.56](https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage) it does all the "Get Started" [...]

@mprowe 
I just added step-2 in above steps
rm ~/.config/BambuStudio/BambuNetworkEngine.conf

please try again

mprowe, 2023-12-16:
I don't have a BambuNetworkEngine.conf
```
mprowe@localhost:~/.config/BambuStudio> ls -al
total 12
drwxr-xr-x 1 mprowe users    86 Dec 16 06:05 .
drwx------ 1 mprowe users  1920 Dec 16 10:04 ..
-rw-r--r-- 1 mprowe users 10140 Dec 16 06:05 BambuStudio.conf
drwxr-xr-x 1 mprowe users    22 Dec 16 05:59 hms
drwxr-xr-x 1 mprowe users    72 Dec 16 05:58 log
drwxr-xr-x 1 mprowe users     0 Dec 16 05:58 ota
drwxr-xr-x 1 mprowe users   256 Dec 16 05:58 printers
drwxr-xr-x 1 mprowe users    22 Dec 16 05:58 system
drwxr-xr-x 1 mprowe users    38 Dec 16 06:05 user
mprowe@localhost:~/.config/BambuStudio> 

```

lanewei120, 2023-12-16:
> I don't have a BambuNetworkEngine.conf
>
> ```
> mprowe@localhost:~/.config/BambuStudio> ls -al
> total 12
> drwxr-xr-x 1 mprowe users    86 Dec 16 06:05 .
> drwx------ 1 mprowe users  1920 Dec 16 10:04 .. [...]

so please help to upload the directory of log here
I will check the details
@mprowe

lanewei120, 2023-12-16:
> I unzipped libbambu_networking.zip into ~/.config/BambuStudio/plugins, but every time I start BambuStudio_linux_ubuntu_v01.08.01.57.AppImage it immediately downloads and installs the original [...]

@klsgh you need to use 1.8.2
please follow the steps I updated above
and check whether it work or not for you

currently these steps are ok for me under openSUSE

klsgh, 2023-12-16:
I started BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage and it did not automatically
install the network plugin (note that I had deleted it from the previous version, so I could
continue to work). I then clicked on the "here" link to install the network plugin, which it did,
and then I closed the app.

I unzipped libbambu_networking.zip into /home/user.name/.config/BambuStudio/plugins and restarted
the app. It immediately downloaded the original network plugin again and overwrote the unzipped
version.

lanewei120, 2023-12-16:
> I started BambuStudio_linux_ubuntu_v01.08.02.56-20231213.AppImage and it did not automatically install the network plugin (note that I had deleted it from the previous version, so I could continue [...]

@klsgh 
could you give the log for us to debug
and after you unzip the libraries, please give a snapshot of the plugins directory with ls -l

mprowe, 2023-12-16:
[debug_Sat_Dec_16_05_58_51_9706.log.0.zip](https://github.com/bambulab/BambuStudio/files/13692374/debug_Sat_Dec_16_05_58_51_9706.log.0.zip)

... and
```
mprowe@localhost:~/.config/BambuStudio/plugins> ls -l
total 7156
-rw-r--r-- 1 mprowe users 7325768 Dec 15 21:14 libbambu_networking.so
mprowe@localhost:~/.config/BambuStudio/plugins> 

```

lanewei120, 2023-12-16:
> [debug_Sat_Dec_16_05_58_51_9706.log.0.zip](https://github.com/bambulab/BambuStudio/files/13692374/debug_Sat_Dec_16_05_58_51_9706.log.0.zip)
>
> ... and
>
> ``` [...]

@mprowe 

seems there are other libraries missed on your machine
[linux_01.08.02.01.zip](https://github.com/bambulab/BambuStudio/files/13692736/linux_01.08.02.01.zip)
you can unzip these libraries into plugins directory and try again

mprowe, 2023-12-16:
Wooo....Hoooo....
Its working. Thank you.

klsgh, 2023-12-16:
[debug_Sat_Dec_16_16_24_58_8294.log.0.zip](https://github.com/bambulab/BambuStudio/files/13693005/debug_Sat_Dec_16_16_24_58_8294.log.0.zip)

kls@eagle:/home/kls/.config/BambuStudio/plugins > ls -l
total 20568
drwxr-xr-x 2 kls users    4096 Dez 16 16:22 backup
-rw-r--r-- 1 kls users 6724640 Dez  7 11:21 libBambuSource.so
-rw-r--r-- 1 kls users 7325768 Dez 15 21:14 libbambu_networking.so
-rw-r--r-- 1 kls users 7001584 Dez  7 11:21 liblive555.so

lanewei120, 2023-12-17:
> [debug_Sat_Dec_16_16_24_58_8294.log.0.zip](https://github.com/bambulab/BambuStudio/files/13693005/debug_Sat_Dec_16_16_24_58_8294.log.0.zip)
>
> kls@eagle:/home/kls/.config/BambuStudio/plugins > ls -l [...]

@klsgh 

dlerror is /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /home/kls/.config/BambuStudio/plugins/libbambu_networking.so

it seems your case is different, these libraries are built on my openSUSE, which use a newer glibc version
your system seems a bit old

I will try to build an old version for you next week in office

lanewei120, 2023-12-18:
@klsgh 

please have a try for these new libraries
[linux_01.08.02.01_new.zip](https://github.com/bambulab/BambuStudio/files/13698513/linux_01.08.02.01_new.zip)

klsgh, 2023-12-18:
@lanewei120 Now it works! Thank you so much!!

quantalume, 2023-12-27:
> @mprowe @klsgh sorry, I didn't give the clear steps before
> 
> please do as follows:
> 

Can confirm that this solution is working for me as well. Thanks!

closed 2024-01-18 by lanewei120