bambulab/BambuStudio #2664: Bambu Studio prevents the display turning off / computer sleeping if you have watch the printer camera video and then stopped it but stayed on the device tab

AdrianGarside, 2023-10-04:
### Bambu Studio Version

1.7.0

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

1. View camera feed from printer
2. Stop the feed
3. Run 'powercfg /requests' in an elevated command prompt

BUG:
Display and System are blocked by bambu studio. The display will never turn off and the machine will never sleep.

4. Navigate away from the device tab and the blocked state will resolve.

### Actual results

Display stays on. Machine never sleeps.

### Expected results

Having stopped the video there is no reason Bambu Studio to force the display to stay on or to block the machine from sleeping.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-10-07:
There is a feature request relate to this.

closed 2023-10-07 by bambu123

AdrianGarside, 2023-10-07:
Do you know which release will have the fix?

AdrianGarside, 2023-10-07:
Oops, jinxed it. This time when I stopped the video playback, I hit the bug:
[image]
Now nothing I do corrects it other than closing bambu studio.

bambu123, 2023-10-07:
We haven't plan to work on the feature. But this is a duplicate issue.
You can post your request here. #2477

AdrianGarside, 2023-10-07:
@bambu123 that issue is complaining about a technically by design behavior - video play back on windows generally should stop the display (and machine) turning off. 

This is complaining about bugs in the current behavior where Bambu Studio often blocks the display turning off or the machine going to sleep permanently having ever viewed (and then stopped) the video preview.

Are you going to be blocking the current behavior completely under #2477?

ArnieO, 2024-06-01:
I have this issue on a new installation. Hoping now that this fix remedies it: https://github.com/bambulab/BambuStudio/issues/2477#issuecomment-1716304349

PunaJussi, 2024-06-06:
This really should be fixed.

bryanhunwardsen, 2024-08-02:
I can confirm this is fixed on current release as of today (at least with a single instance of bambu studio open and connected to a single/active machine. I checked the power state lock before , during and after preview is stopped and bamabu studio cleaned the lock up after video stream stopped.

There may be some edge cases (I have logged in one of the linked issues) that orphan the lock but I can verify for Windows 10, the simple case of removing the lock when stream is stopped is working.