bambulab/BambuStudio #2665: fatal error, exception catched: invalid string position

jcasey763, 2023-10-04:
### Bambu Studio Version

1.7.6

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

amd ryzen 5 3600 6-core processor

### Printer

bambu lab p1p

### How to reproduce

1. download software
2. finish installation
3. attempt to open software

### Actual results

received error code stating "fatal error, exception catched: invalid string position"

### Expected results

after following above steps I should be able to access the software

### Project file & Debug log uploads

system will not let me copy error message direct from box and cannot open application to get files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-10-07:
Could you provided the log file for us to check？We can not reproduce this issue under our PC environment.
[image]

zk213, 2023-10-11:
Hey, i also met this problem just after yesterday's update
[debug_Wed_Oct_11_18_06_10_29456.log.zip](https://github.com/bambulab/BambuStudio/files/12868081/debug_Wed_Oct_11_18_06_10_29456.log.zip)

jcasey763, 2023-11-08:
> What did you do to fix this I have it right now

It randomly started working after a previous update but I used orcaslicer
in place of it and had no issues with that.

Marsian999, 2024-01-18:
[image]
i having that same problem after the lastest update
fatal error, exception catched: invalid string position

Marsian999, 2024-01-18:
[image]
sorry wrong pic

Marsian999, 2024-01-18:
[image]
i also having the same problem
bambu studio GUI initialization failed
fatal error, exception catched: invalid string position

lanewei120, 2024-01-19:
> ![Screenshot 2024-01-18 141449](https://private-user-images.githubusercontent.com/49322933/297856245-0e200c54-f9e9-4a2c-8f81-b10170a081c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MjU1MzYsIm5iZiI6MTcwNTYyNTIzNiwicGF0aCI6Ii80OTMyMjkzMy8yOTc4NTYyNDUtMGUyMDBjNTQtZjllOS00YTJjLThmODEtYjEwMTcwYTA4MWM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE5VDAwNDcxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxMjI4ZGZmMTFhOGZkOTMxYTEwNGJmMTViZGQ0MGE4MTgxYzIyYTJlMDI0NTE0ZjhhNTJkOGM4N2VhNWFkYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3QbM11RYRmBtOvj2xoZuU5cgt-4Od6xD_ZQvyLVETkc) i also having the same problem bambu studio GUI initialization failed fatal error, exception catched: invalid string position

usually, you just need to remove BambuStudio.conf and BambuNetworkEngine.conf
and reboot the software

closed 2024-06-26 by lanewei120

ClickDaHouseCat, 2024-11-06:
I resolve this problem. I uninstalled with 'Delete user data" and installed again.

TOOKWEAR, 2025-12-12:
nope i did it after restart of bambustudio again this