bambulab/BambuStudio #2667: Add shape from SVG file similar to how you currently have the Text feature

sprior, 2023-10-05:
When you added the Add Text feature which was really cool I immediately thought an obvious next step would be to add a shape based on a SVG file.  It would be basically the same implementation as the add Text you've got now except that instead of a string in a font you'd take a SVG file as the 2D shape to start from.

I think this would be useful as an addition to the "Add primitive" menu as the first shape to be added to a build plate as well as a decoration on top of an existing model as you currently do the "Add Text".

This would allow people to add logos on top of an existing model, or someone with no CAD knowledge at all to add a base shape based on a SVG file and then add text on top of it to create a keychain or something similar very easily.  I think once this is added someone could build up an object by layering SVG shapes one on top of the other and giving each one a thickness.

I'm only suggesting this because it seems like it would be easy to extend the add text implementation code to use the additional source of 2D shape.

zhimin-zeng-bambulab, 2023-10-11:
This is a good suggestion and we will improve it in the future

sprior, 2023-10-12:
[image]

Just as an example, for Halloween my wife has been decorating our front stairs with cardboard mouse silhouette and they were getting worn so I found some SVG clip art files, imported them into CAD, extruded 0.6mm, exported as STLs, then sliced (playing around with X and Y scaling) and printed.  Would have been so much more convenient if I could have just imported the SVG into the slicer with a thickness.

zhimin-zeng-bambulab, 2023-10-12:
@sprior  You can import svg files in Studio

sprior, 2023-10-12:
@zhimin-zeng-bambulab Oh that's great, thanks!  The feature to add an SVG like you currently add text would still be a nice addition.

binaryanomaly, 2023-10-14:
@zhimin-zeng-bambulab Please have a look at UltiMaker Cura. It can import even more file types and create a usable results out of it https://support.makerbot.com/s/article/1667411286871 - looks better to me than what bambu currently does out of .svg's.

Ideally also the coloring would be applied according to the original as part of an import.
This would be a fantastic feature and would enable very easy and quick 3d printing of 2d images without much fiddling as it is required today.

FickingGames, 2023-12-07:
I believe PrusaSlicer 2.7 just released this feature. This would help my workflow tremendously.