bambulab/BambuStudio #2668: Layer shifts occurring with filament swap (ams)

Ac3inSpac3, 2023-10-05:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S with AMS

### How to reproduce

When printing with ams multi filament (pla support and pla) the print head gets shifted across roughly 1mm to the left and 1mm to the front of the print bed. Seems to happen with every filament change. Seems that the print head goes back to the wrong position every purge on filament swap. 
same print comes out fine with no filament change (see last 2 images)
Seems to have occurred after recent printer update -> version 01.04.01.00

### Actual results

[image]
[image]
[image]
[image]

All printed with PLA (No support PLA)
[image]
[image]

### Expected results

[image]
[image]
[image]

### Project file & Debug log uploads

[layerShiftFile.zip](https://github.com/bambulab/BambuStudio/files/12812415/layerShiftFile.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-10-07:
Can not find any abnormal point in gcode. Need to be checked by printer firmware team.

pichen2021, 2023-10-07:
Hi @Ac3inSpac3 
This issue may be caused by the tool head being stuck for a short time during the filament change process.
Please power off the machine first, [re-tension the belt](https://wiki.bambulab.com/en/x1/maintenance/belt-tension) and then manually move the tool head to the four corners and check whether the PTFE tube or the TYPEC cable will affect the movement of the tool head, if the issue still exists, we encourage you to export the log(+a video of the tool head movement when this issue happed would be better), reach out to our customer service team [Click here to open a new ticket in our Support Page](https://bambulab.com/en/my/support/tickets)
Thanks!

rsmilne7, 2023-10-12:
Also having this same issue, I can confirm this wasn't always happening. Happens on only the first pause/color change for me. All other pauses or color changes seem to work ok.
firmware version 01.04.01.00

martinmalek, 2023-12-20:
Bumping this. AMS did a filament change (A4 to A3) after one of the rolls was finished after which I noticed a shift in 

Model: X1C
Firmware version: 01.07.00.00
AMS version: 00.00.06.40

Print is still ongoing but I screenshot two pictures from the camera.

[image]
[image]