bambulab/BambuStudio #2671: Precision/Resolution should be applied before Fuzzy Skin is applied

DzzD, 2023-10-05:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1 Carbone

### How to reproduce

1 - Enable Fuzzy Skin
2 - Set Fuzzy thickness to a low value like 0.05
3 - Set Precision=>résolution to 0.04

### Actual results

Fuzzy skin effect almost disappear everywhere

### Expected results

Fuzzy skin effect should not disappear.

Being able to set precision option to a little higher value is a very neat option, but unfortunately it is fighting against fuzzy skin effect and so both are incompatible if using a low value for Fuzzy Skin thickness option.

Applying Fuzzy Skin effect after resolution algorithme rather than before should resolve this bug.

### Project file & Debug log uploads

[PlanerVacumConnector.3mf.zip](https://github.com/bambulab/BambuStudio/files/12820059/PlanerVacumConnector.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-10-07:
[image]
This setting is gcode resolution and be applied during exporting gcode. Bacause arc fitting is also done in this step, and clipper lib doesn't support 2D boolean of arc segment, this step can not be put before fuzzy skin.

This setting also influence the slicing resolution when slicing 3D to 2D polygon, which is before fuzzy skin.

We can separate this setting into two setting: slicing resolution and gcode resolution. Slicing resolution can be applied before fuzzy skin, and the gcode resolution must be after. 

If set Fuzzy thickness to a low value like 0.05, which  is large than gcode resolution, fuzzy skin effect will still disappear.

SaltWei, 2023-10-07:
> Applying Fuzzy Skin effect after resolution algorithme rather than before should resolve this bug.

So this is impossible.

Another method for this issue if slicer automatically limit the gcode resolution not larger than fuzzy thickness if user set non-zero value of fuzzy thickness. This can be automatically handled when slicing.

DzzD, 2023-10-10:
>We can separate this setting into two setting: slicing resolution and gcode resolution. Slicing resolution can be applied before >fuzzy skin, and the gcode resolution must be after.

This seems to be an elegant and versatile solution