bambulab/BambuStudio #2675: Studio cloud presets corrupted, cannot delete them

s-kimball, 2023-10-07:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Open Bambu Studio
Change printer in ‘prepare’ tab away from P1S default to observe that Studio thinks it has been modified.
See following message:
[image]
If attempt to print anyway, receive this message:
[image]

### Actual results

Bambu Studio thinks the default profile for the P1S is modified and for another printer. This is due to corrupted file on the cloud being automatically downloaded.  There's no way to delete or remove the default profiles in the cloud so I am unable to use the cloud anymore.

### Expected results

Default profiles should **never** come from a cloud, it should be baked into Bambu Studio and **resettable** to avoid these issues. The default profile should be the correct profile. 

### Project file & Debug log uploads

[debug_Sat_Oct_07_08_21_41_9284.log.zip](https://github.com/bambulab/BambuStudio/files/12837849/debug_Sat_Oct_07_08_21_41_9284.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

smkimbal, 2023-10-17:
Follow-up: resolution was found by deselecting printers with this issue (in my case P1S), unchecking auto-sync in preferences, and reselecting default profiles so they're correctly installed via Studio. At this point you cannot autosync, it will still download from the cloud. Instead if you wait long enough the local files will be deemed "newer" and overwrite profiles in the cloud when reselecting autosync. I think this window is 10 minutes, but I waited a full day.

bambu123, 2023-10-24:
> Follow-up: resolution was found by deselecting printers with this issue (in my case P1S), unchecking auto-sync in preferences, and reselecting default profiles so they're correctly installed via [...]

We may upgrade the system preset from the cloud, and the upgrade requires user confirmation before proceeding. Are these system presets the default profiles you mentioned?

bambu123, 2023-11-07:
> Change printer in ‘prepare’ tab away from P1S default to observe that Studio thinks it has been modified.

If you think you haven't manually modified these parameters, you can directly discard these modifications. These modifications may come from open project files.