bambulab/BambuStudio #2678: Could you please make the AMS Cabin humidity and Desiccant status indicator a bit bigger?

kennethfinn, 2023-10-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
**_The AMS Cabin humidity and Desiccant status indicator is very very Tiny_**

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
_**Can you please make it bigger?**_

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kennethfinn, 2023-10-07:
Duplicate issue. already reported by someone else

closed 2023-10-07 by kennethfinn

DanTup, 2024-04-06:
I couldn't find a dupe of this except one that was already implemented (https://github.com/bambulab/BambuStudio/issues/1120), but it's still very unreadable IMO. I filed https://github.com/bambulab/BambuStudio/issues/3827 as a request to make it larger and/or include in the popup.